bitbank2/AnimatedGIF #1: /homer.gif  where?

leftCoast, 2020-07-28:
Sorry if this is the wrong place to ask a question. (Kinda' new to github, if confuses me a lot.)

Where can I find the actual file /homer.gif that the example for the SD drive is looking for?

-jim lee

bitbank2, 2020-07-28:
The example images are stored in the /test_images directory. If you're reading files from an SD card, then you would need to copy it there first. If including it in your sketch to be written to FLASH memory, then do:

#include "../test_images/homer.h"

closed 2020-07-28 by bitbank2

leftCoast, 2020-07-28:
I was thinking the .gif file would be in the examples, so that I could copy it to me SD card.

leftCoast, 2020-07-28:
Had some issues, using a adafruit 1947 tied to a teensy 3.2. Had to
Initialze the display before starting the SD drive because if you don't
the screen can (will)  mess up communication to the SCD drive.

I think I have the thing running, but nothing is showing up on the screen.
The GIFReadFile() is grabbing 255 bytes then 1 byte then 255 bytes over
and over. But no drawing.

I'll attach the file I'm using as my example GIF.

-jim lee

bitbank2, 2020-07-28:
The file read is happening as it must with GIF files because of the way the data is stored. I can't see your sketch, so I can't see what you're doing wrong.

leftCoast, 2020-07-28:
On Tue, July 28, 2020 4:46 am, Larry Bank wrote:
> The file read is happening as it must with GIF files because of the way
> the data is stored. I can't see your sketch, so I can't see what you're
> doing wrong.

Oh sorry, here's the .ino file

#include <AnimatedGIF.h>

#include "SPI.h"
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"
#include <SD.h>

#define TFT_CS         10
#define TFT_RST        8
#define TFT_DC         9
#define TFT_MOSI       11
#define TFT_CLK        13
#define TFT_MISO       12
#define BUILTIN_SDCARD 4

Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK,
TFT_RST, TFT_MISO);
AnimatedGIF gif;
File f;

void * GIFOpenFile(char *fname, int32_t *pSize)
{
   //Serial.println("In callback?");Serial.flush();
   //Serial.print("fname : ");Serial.println(fname);Serial.flush();
  f = SD.open(fname);
  //Serial.println("We make it past open()?");Serial.flush();
  if (f)
  {
    //Serial.println("We got an f..");Serial.flush();
    *pSize = f.size();
    return (void *)&f;
  }
  //Serial.println("No f !?");Serial.flush();
  return NULL;
} /* GIFOpenFile() */

void GIFCloseFile(void *pHandle)
{
  File *f = static_cast<File *>(pHandle);
  if (f != NULL)
     f->close();
} /* GIFCloseFile() */

int32_t GIFReadFile(GIFFILE *pFile, uint8_t *pBuf, int32_t iLen)
{
    int32_t iBytesRead;
    Serial.print("Looking to read : ");Serial.print(iLen);Serial.println("
bytes");
    iBytesRead = iLen;
    File *f = static_cast<File *>(pFile->fHandle);
    // Note: If you read a file all the way to the last byte, seek() stops
working
    if ((pFile->iSize - pFile->iPos) < iLen)
       iBytesRead = pFile->iSize - pFile->iPos - 1; // <-- ugly work-around
    if (iBytesRead <= 0)
       return 0;
    iBytesRead = (int32_t)f->read(pBuf, iBytesRead);
    pFile->iPos = f->position();
    Serial.print("read out : ");Serial.print(iBytesRead);Serial.println("
bytes");
    return iBytesRead;
} /* GIFReadFile() */

int32_t GIFSeekFile(GIFFILE *pFile, int32_t iPosition)
{
  int i = micros();
  File *f = static_cast<File *>(pFile->fHandle);
  f->seek(iPosition);
  pFile->iPos = (int32_t)f->position();
  i = micros() - i;
  Serial.printf("Seek time = %d us\n", i);
  return pFile->iPos;
} /* GIFSeekFile() */

// Draw a line of image directly on the LCD
void GIFDraw(GIFDRAW *pDraw)
{
    uint8_t *s;
    uint16_t *d, *usPalette, usTemp[320];
    int x, y;

    usPalette = pDraw->pPalette;
    y = pDraw->iY + pDraw->y; // current line

    s = pDraw->pPixels;
    // Apply the new pixels to the main image
    if (pDraw->ucHasTransparency) // if transparency used
    {
      uint8_t *pEnd, c, ucTransparent = pDraw->ucTransparent;
      int x, iCount;
      pEnd = s + pDraw->iWidth;
      x = 0;
      iCount = 0; // count non-transparent pixels
      while(x < pDraw->iWidth)
      {
        c = ucTransparent-1;
        d = usTemp;
        while (c != ucTransparent && s < pEnd)
        {
          c = *s++;
          if (c == ucTransparent) // done, stop
          {
            s--; // back up to treat it like transparent
          }
          else // opaque
          {
             *d++ = usPalette[c];
             iCount++;
          }
        } // while looking for opaque pixels
        if (iCount) // any opaque pixels?
        {
          tft.setAddrWindow(pDraw->iX+x, y, iCount, 1);
          tft.startWrite();
          tft.writePixels(usTemp, iCount, false, false);
          tft.endWrite();
          x += iCount;
          iCount = 0;
        }
        // no, look for a run of transparent pixels
        c = ucTransparent;
        while (c == ucTransparent && s < pEnd)
        {
          c = *s++;
          if (c == ucTransparent)
             iCount++;
          else
             s--;
        }
        if (iCount)
        {
          x += iCount; // skip these
          iCount = 0;
        }
      }
    }
    else
    {
      s = pDraw->pPixels;
      // Translate the 8-bit pixels through the RGB565 palette (already
byte reversed)
      for (x=0; x<pDraw->iWidth; x++)
        usTemp[x] = usPalette[*s++];
      tft.setAddrWindow(pDraw->iX, y, pDraw->iWidth, 1);
      tft.startWrite();
      tft.writePixels(usTemp, pDraw->iWidth, false, false);
      tft.endWrite();
    }
} /* GIFDraw() */

void setup() {
  Serial.begin(115200);
  while (!Serial);

// Note - some systems (ESP32?) require an SPI.begin() before calling
SD.begin()
// this code was tested on a Teensy 4.1 board
   tft.begin();
   tft.fillScreen(ILI9341_BLACK);

  if(!SD.begin(BUILTIN_SDCARD))
  {
    Serial.println("SD Card mount failed!");
    return;
  }
  else
  {
    Serial.println("SD Card mount succeeded!");
  }

  // put your setup code here, to run once:

  gif.begin(LITTLE_ENDIAN_PIXELS);
}

void loop() {
  // put your main code here, to run repeatedly:
  //Serial.println("About to call gif.open");
  if (gif.open((char *)"burn.gif", GIFOpenFile, GIFCloseFile, GIFReadFile,
GIFSeekFile, GIFDraw))
  {
    Serial.printf("Successfully opened GIF; Canvas size = %d x %d\n",
gif.getCanvasWidth(), gif.getCanvasHeight());Serial.flush();
    while (gif.playFrame(true, NULL))
    {
    }
    gif.close();
  }
  else
  {
    //Serial.println("Error opening file");Serial.flush();
    while (1)
    {};
  }
}

bitbank2, 2020-07-28:
You shouldn't call SD.begin() twice.
The filename you passed to gif.open, "burn.gif" probably won't work. You need to pass a complete pathname. If the file resides in the root directory of the SD card, then it should be "/burn.gif".

The rest looks ok if you got all of the pin numbers correct.

leftCoast, 2020-07-28:
What looks like an SD.begin() is actually a comment. Added the slash as the leading char of the pathname.

Here's what I get as output on the serial monitor.

`Seek time = 2 us
Successfully opened GIF; Canvas size = 200 x 109
Seek time = 2 us
Successfully opened GIF; Canvas size = 200 x 109
Seek time = 2 us
Successfully opened GIF; Canvas size = 200 x 109`

-jim lee

bitbank2, 2020-07-28:
That looks correct; if it's not drawing on the display, try drawing some text before calling the gif class to make sure you set up the display properly.

leftCoast, 2020-07-29:
Yup, draws text just where I tell it to, and even the color I chose.

The hardware is a handheld programmer I use for another project so it's pretty well had the bugs beat out of it.

-jim lee

bitbank2, 2020-07-29:
I can't debug it without more to go on. The code looks ok if your display is connected correctly.

leftCoast, 2020-07-29:
Did you try displaying the .gif I sent you? Maybe that's the issue? The code is what I got from your library as an example. The only real change was initializing the display before the SD drive and the .gif file I"m trying to view. Do you have a know good .gif file I could try on this end?

bitbank2, 2020-07-29:
I didn't receive a gif from you. Please email it to bitbank@pobox.com