bitbank2/AnimatedGIF #2: Some problem GIFs

PaintYourDragon, 2020-07-28:
This is looking super encouraging! Wow. With some tweaks, on PyPortal it’s able to play back a full-screen 200 MB GIF (Big Buck Bunny) at full speed (I’ll put the GIF on DropBox if you’re interested). Adafruit_Arcada_GifDecoder can’t quite keep up with that one. Looks like about a 10% boost.

A few GIFs are having trouble. They’re in this branch, which includes the PyPortal example:
https://github.com/PaintYourDragon/AnimatedGIF/tree/pb-experiments
In the 'gifs' folder:

- beast.gif has issues with transparency and frame timing (view in web browser to compare). It’s close but looks like something’s off by one. Adafruit_Arcada_GifDecoder gets the transparency right but still has timing trouble with that one.
- dragons.gif doesn’t display anything. It’s doing the frame loop, but displaying nothing and eventually hangs (not sure if it’s the last frame/EOF or in the middle somewhere).
- krampus-anim.gif has issues similar to dragons.gif. Adafruit_Arcada_GifDecoder sorta plays this one but has trouble with the transparency (again, compare against browser).

Also in that fork/branch, small change in AnimatedGIF.h sets MAX_WIDTH to 480 on PyPortal Titano, 320 otherwise. PyPortal example uses SdFat (instead of SD library), gets a noticeable boost from that, and also does some big-endian DMA stuff to the screen.

bitbank2, 2020-07-28:
Thanks for testing it. I don't have any Adafruit ATSAMD51 hardware on hand to test with (they're in Rome and I'm in Florida). I'll check out the problems later today.
I have the BigBuckBunny file - mine is 320x180 and 267MB. I tested it on the Teensy 4.1 

I have a feeling the speed is more than 10% faster than Adafruit's, but the display output is holding it back 😄

bitbank2, 2020-07-28:
P.S. I 'carefully crafted' the file access logic to avoid seeking on the SD card. This is the cause of major slowdowns. Reading from the SD card is not super fast, but seeking can take ridiculous amounts of time.

closed 2020-07-28 by bitbank2