bitbank2/AnimatedGIF #5: Add an option to not loop whole gif

BorisFR, 2020-08-16:
I'm doing a "clock" with an ESP32 and Matrix Panel.
I don't want to cut animation by duration, I just want that they play only once.

This is what I have add to your code:
**AnimatedGIF.h**

    public:
      void enableLoop(bool enable);

    private:
      bool loopEnable = true;

**AnimatedGIF.cpp**
in PlayFrame()

    if (_gif.GIFFile.iPos >= _gif.GIFFile.iSize - 1) // no more data exists
    {
        if (!loopEnable)
            return 0;
        (*_gif.pfnSeek)(&_gif.GIFFile, 0); // seek to start
    }

If you think it is ok, it will be cool for me if you add it to the library. But I can understand that you do not add my modifications to your code.

Thanks for your library!

bitbank2, 2020-08-16:
The code for playFrame() already returns a 0 when you hit the last frame. I don't understand what your suggestion adds.
You can use the reset() if you want to play it again without re-opening the file.

BorisFR, 2020-08-16:
So I've got a problem with that file that will never ending cause I never receive 0.
[image]

bitbank2, 2020-08-16:
thanks for providing a sample image. I'll see if I can fix it.

bitbank2, 2020-08-16:
I just pushed a fix, please give it a try.

closed 2020-08-16 by bitbank2

BorisFR, 2020-08-17:
Great! Just test and now it is ok, cool!.
It also solve another problem I have on another gif that stop displaying at a moment. Now, this one play fine also.

It's ok for me!
Thanks a lot.

bitbank2, 2020-08-17:
Glad it worked and thank you for providing the sample image. If I can recreate the problem, I can solve it 😄