bitbank2/AnimatedGIF #7: Some GIFs have wrong timing animation

BorisFR, 2020-08-17:
When using the library, they are too slow.
The common thing I found is that playFrame() tell me to wait 90 or 91 ms for that GIFs. 
[image]
[image]
[image]

bitbank2, 2020-08-17:
When an image provides 0 as the frame delay, I return 100ms (10 FPS) because I believe that's the default for browsers. I'll take a look at your images.

BorisFR, 2020-08-17:
I think you're right.
If I received 100 I then change to 10 and the animation are ok. I have try with 16 but it is still too low. 10 seems to be ok but I don't know the real standard value. Or perhaps it exist a default value in the gif format. My level is not sufficient to help more.

bitbank2, 2020-08-18:
I pushed a change which allows animation delays of 20ms or more. Please re-test the latest code.

closed 2020-08-18 by bitbank2

BorisFR, 2020-08-27:
Sorry for the delay: just test and it is now ok. Thank you.