bitbank2/AnimatedGIF #12: [IDEA] - Trigger the advance of the gif with an external event

FrenchLab47, 2020-09-13:
Hello bitbank2,

I am continuing my experiments with your library.
I noticed small position defects with "optimized" transparent gifs.
But everything is fine if I remove the optimization with software like Gif Movie Gear

Now I would like to synchronize or trigger the frames of the gif, one by one, with external events.

Some kind of function like this:
`playFrameNumber (int frameNB) {…} `
or frameNB would be the number of my choice, which I could increment in a loop with other graphics.

This function would allow for example:
- Display a Gif image with a clock, and change frames every second.
- Or a Gif thermometer which changes according to the values received.
- Or move an object or a text followed by a character (gif) walking at the same speed.
 
So far I have just tried it with your normal playFrame function.
I canceled the delay (with bSync = false) and tried to pass two more values:
`int AnimatedGIF :: playFrame (bool bSync, int * delayMilliseconds, int curFrameNb, int frameToPlay) {…}`
   
This way I can scroll through all the frames quickly and get out of the loop to block the requested frame.
But fast frames are also displayed for a fraction of a second, and that causes artefacts.

The ideal would be to advance the file to the correct position without displaying the previous frames, or displaying them in black (color 0).

If you have an idea that is more "elegant" than mine, I am listening.
Thank you for your work.

bitbank2, 2020-09-13:
Hi FrenchLab47,
I understand your idea, but what you're asking for is best left up to the user (you). You cannot skip GIF frames by decoding them and not displaying the new pixels because each frame builds the image in sequence. Your output would be incorrect for all files except ones that you create yourself which don't depend on the previous frames. It would certainly be faster to skip the output stage inside my decoder if I knew that the pixels didn't need to be displayed, but this would be a feature just for you. If you have independent GIF images then put them in separate files (e.g. clock digits) and work with them one at a time. This way you won't need to "skip" any images in a sequence.

closed 2020-09-13 by bitbank2

FrenchLab47, 2020-09-14:
Yes indeed, I will organize my gifs with a single frame, it will do less calculations.
I already do this to display a fixed logo and I suspect it is the easiest.
Thank you for your reply.

embedded-creations, 2020-11-04:
@FrenchLab47 Adafruit has a project that does this, but I imagine it works only with certain GIFs.  They do it inside the Arduino sketch.  I _think_ this might work with the AnimatedGIF library too, but I haven't tried it out.

https://learn.adafruit.com/gaming-handheld-crank/code

https://github.com/adafruit/Adafruit_Arcada_GifDecoder/blob/master/examples/crankyGIFs/crankyGIFs.ino#L173

If you try it, can you let me know?

bitbank2, 2020-11-04:
I don't have time to try it now, but it looks like it should work with my code the same way. The only danger is that you would need to seek to a "key" frame and not one that depends on previous pixels.

FrenchLab47, 2020-11-04:
Hello,
Thank you for your information.
Finally I managed to do exactly what I want with the code of bitbank2 without modifying it. I am using global variables (xmove, ymove and frameNB). My gifs move by walking with the text. Or it change with external events. I can also make clocks with big numbers in transparent gif on background images.
I will release the code for my led display soon, but I still have some details to work out (image server, OTA update, Android application, and a lot of cleaning work).