bitbank2/AnimatedGIF #13: gif.open argument error

MeCK-Frog, 2020-09-14:
The following error occurs and it cannot be executed.
The gif.open method seems to be in error, do you know the cause?
thank you.

Error statement
=====================================================================================
In file included from /Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3/src/M5Stack.h:122:0,
                 from /Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3/src/ESP32-Chimera-Core.h:2,
                 from /Users/*****/Documents/Arduino/AnimatedGIF/examples/ESP32-LGFX-SDCard-GifPlayer/ESP32-LGFX-SDCard-GifPlayer.ino:1:
/Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3/src/M5Display.h: In member function 'void M5Display::writePixels(uint16_t*, uint32_t)':
/Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3/src/M5Display.h:50:35: warning: 'void lgfx::LGFXBase::pushColors(T*, int32_t, bool) [with T = short unsigned int; int32_t = int]' is deprecated: use pushPixels [-Wdeprecated-declarations]
       pushColors(colors, len, true);
                                   ^
In file included from /Users/*****/Documents/Arduino/libraries/LovyanGFX/src/LovyanGFX.hpp:43:0,
                 from /Users/*****/Documents/Arduino/libraries/LovyanGFX/src/LGFX_TFT_eSPI.hpp:14,
                 from /Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3/src/M5Display.h:10,
                 from /Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3/src/M5Stack.h:122,
                 from /Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3/src/ESP32-Chimera-Core.h:2,
                 from /Users/*****/Documents/Arduino/AnimatedGIF/examples/ESP32-LGFX-SDCard-GifPlayer/ESP32-LGFX-SDCard-GifPlayer.ino:1:
/Users/*****/Documents/Arduino/libraries/LovyanGFX/src/lgfx/LGFXBase.hpp:481:43: note: declared here
     [[deprecated("use pushPixels")]] void pushColors(T*                   data, std::int32_t len, bool swap) { startWrite(); writePixels(data, len, swap); endWrite(); }
                                           ^
/Users/*****/Documents/Arduino/AnimatedGIF/examples/ESP32-LGFX-SDCard-GifPlayer/ESP32-LGFX-SDCard-GifPlayer.ino: In function 'int gifPlay(char*)':
ESP32-LGFX-SDCard-GifPlayer:162:89: error: invalid conversion from 'void* (*)(char*, int32_t*) {aka void* (*)(char*, int*)}' to 'void* (*)(const char*, int32_t*) {aka void* (*)(const char*, int*)}' [-fpermissive]
   if( ! gif.open( gifPath, GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile, GIFDraw ) ) {
                                                                                         ^
In file included from /Users/*****/Documents/Arduino/AnimatedGIF/examples/ESP32-LGFX-SDCard-GifPlayer/ESP32-LGFX-SDCard-GifPlayer.ino:10:0:
/var/folders/n5/ddy9p4_d5k75mfsktms5jvhw0000gp/T/arduino_build_644108/sketch/AnimatedGIF.h:138:9: note:   initializing argument 2 of 'int AnimatedGIF::open(const char*, void* (*)(const char*, int32_t*), void (*)(void*), int32_t (*)(GIFFILE*, uint8_t*, int32_t), int32_t (*)(GIFFILE*, int32_t), void (*)(GIFDRAW*))'
     int open(const char *szFilename, GIF_OPEN_CALLBACK *pfnOpen, GIF_CLOSE_CALLBACK *pfnClose, GIF_READ_CALLBACK *pfnRead, GIF_SEEK_CALLBACK *pfnSeek, GIF_DRAW_CALLBACK *pfnDraw);
         ^
/Users/*****/Documents/Arduino/AnimatedGIF/examples/ESP32-LGFX-SDCard-GifPlayer/ESP32-LGFX-SDCard-GifPlayer.ino: At global scope:
/Users/*****/Documents/Arduino/AnimatedGIF/examples/ESP32-LGFX-SDCard-GifPlayer/ESP32-LGFX-SDCard-GifPlayer.ino:35:13: warning: 'void MyCustomDelay(long unsigned int)' defined but not used [-Wunused-function]
 static void MyCustomDelay( unsigned long ms ) {
             ^
「SD.h」に対して複数のライブラリが見つかりました
使用済：/Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD
未使用：/private/var/folders/n5/ddy9p4_d5k75mfsktms5jvhw0000gp/T/AppTranslocation/7C0A7034-DF5A-4B87-9117-0AE54FB0BF2B/d/Arduino.app/Contents/Java/libraries/SD
次のフォルダのライブラリESP32_Chimera_Core-1.0.3バージョン1.0.3を使用中：/Users/*****/Documents/Arduino/libraries/ESP32_Chimera_Core-1.0.3 
次のフォルダのライブラリWireバージョン1.0.1を使用中：/Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire 
次のフォルダのライブラリSPIバージョン1.0を使用中：/Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI 
次のフォルダのライブラリFSバージョン1.0を使用中：/Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS 
次のフォルダのライブラリSD_MMCバージョン1.0を使用中：/Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD_MMC 
次のフォルダのライブラリSPIFFSバージョン1.0を使用中：/Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPIFFS 
次のフォルダのライブラリSDバージョン1.0.5を使用中：/Users/*****/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SD 
次のフォルダのライブラリLovyanGFXバージョン0.2.1を使用中：/Users/*****/Documents/Arduino/libraries/LovyanGFX 
exit status 1
invalid conversion from 'void* (*)(char*, int32_t*) {aka void* (*)(char*, int*)}' to 'void* (*)(const char*, int32_t*) {aka void* (*)(const char*, int*)}' [-fpermissive]

tobozo, 2020-09-15:
hey @MeCK-Frog thanks for your feedback

looks like a `const` was missing in the `GIFOpenFile` signature : 

`static void * GIFOpenFile(const char *fname, int32_t *pSize)`

I've updated the example sketch, please let me know if this problem persists.

MeCK-Frog, 2020-09-15:
to tobozo
Thank you! 
I was able to compile successfully with your advice.
Another question I would like to ask about the wiring of the SD card reader module.
Insert the SD card into the ESP32 in the readme of your SD_Content file.
Is written.
What should I do to connect the SD card module (HW125)?
And can I ask this question here?
Thank you.

tobozo, 2020-09-15:
@MeCK-Frog is that a built-in or an external SD Card module ? HW125 isn't ringing any bell anyway.

if it's an internal module all you need to know is the pin number for 'the CS pin (can be called "SD_CS" or "TFCARD_CS") and use it with SD.begin( SD_CS ), usually this is pin 4 but your mileage may vary.

tobozo, 2020-09-16:
closing this issue as solved, let's discuss the SD pinout questions on ESP32's [gitter](https://gitter.im/espressif/arduino-esp32)

closed 2020-09-16 by tobozo

MeCK-Frog, 2020-09-16:
tobozo!
I appreciate your cooperation!

MeCK-Frog, 2020-09-16:
thank you very much