bitbank2/AnimatedGIF #14: Some Problem GIFs

PaintYourDragon, 2020-10-30:
Working with the lib on Matrix Portal (RGB LED matrix), several small GIFs were having issues either with display glitches or altogether crashes. Wrestled with the Matrix-side code for a while before just trying the same GIFs over on PyPortal, where the player is known to work. Same issues there, so I’m wondering if this is a problem in AnimatedGIF. The images are unusually small (64x32 pixels) if that’s any indication. These all play fine on desktop system / in browser.

Juggler: glitches.
Lightspeed: crash; no frames displayed.
Werewolf: crash; initial frame displayed but nothing further.

These all look a little dark in browser because they’re manually gamma corrected for the way the LED matrix works…this is normal and known and shouldn’t affect anything.

[image]
[image]
[image]

bitbank2, 2020-10-30:
Thanks for the letting me know. Famous quote - "If I can reproduce it, I can fix it" 😄

bitbank2, 2020-10-30:
Some info... Not a crash, but a normally returned error. The top and bottom images are stored as interlaced and my library rejects them because this code assumes your target CPU doesn't have enough RAM to hold the decoded image. The middle image seems to play correctly.

closed 2020-10-30 by bitbank2

PaintYourDragon, 2020-10-31:
Appreciate your help with this today. Got things _mostly_ resolved by de-interlacing all the GIFs through ImageMagick, and handling the different return values from playFrame().

Two images are still causing issues though. (At least they’re not crashing now, thanks to your playFrame() help…but they do get skipped over in the player now when it detects a -1 value, even though they appear valid on desktop.) Attaching them here. These are not necessarily the same as above (even though they look the same in browser)…they’ve been coalesced & deinterlaced through ImageMagick.

[image]
[image]

bitbank2, 2020-11-01:
On the Mac, both files work correctly; I'll test it on an Arduino board tomorrow. I recently did some fixes to the library, are you sure you're running the latest version?

bitbank2, 2020-11-02:
I just tested both of your updated files on a TTGO T-Display (ESP32 + ST7789 240x135 LCD) and they both play perfectly.

embedded-creations, 2021-02-08:
FYI, the latest version of AnimatedGIF has support for interlaced GIFs (thanks Larry!)

bitbank2, 2021-02-08:
Thanks for helping to test the code :). I should have had interlaced support in from the beginning. I had confused the PNG interlacing with GIF and thought it would break my line-at-a-time code, but it was drop dead simple to add.