bitbank2/AnimatedGIF #15: rgb24 Palette support

embedded-creations, 2020-11-04:
Hi Larry, I've been maintaining Craig Lindley's [AnimatedGIFs sketch](https://github.com/pixelmatix/AnimatedGIFs) for years, and had always wanted to turn it into a platform-independent GIF Decoder library.  I finally got a chance to start [this year](https://www.arduinolibraries.info/libraries/gif-decoder), a few weeks before you released your library and made my work obsolete :-)

I'd like to start using your library for my GIF projects, but it's missing one critical feature: rgb24 palette support.  I'm using GIFs to drive LED matrix displays with high color depth, for projects like this, so I need to get the full 24-bits of color from the source image:

https://www.instructables.com/Continuum-Slow-Motion-LED-Art-Display/

It looks trivial enough to fork your library and change to using rgb24 palettes, but I'd prefer to have the palette type be selectable in the sketch (not through changing a `#define` in the library), without slowing existing rgb565 performance, so other people can use that feature easily.  The best option I can come up with right now is to use C++ templates to choose between the two palette types, but I saw you rewrote your library to use C, so that's probably out.  Arduino doesn't make this kind of library configuration easy.

Do you have any easy solution for this off the top of your head?

Would you even want to support rgb24 in addition to rgb565 in your library, or is this something I should tackle on my own?

I've been enjoying your updates on twitter with your different display libraries.  I don't have any active E-Ink projects, but I do have some in mind that could make use of the work you're doing, keep it up!

bitbank2, 2020-11-04:
That's a reasonable request and easy to implement. Just like the pixel type, I can allow you to choose the palette type before calling open. I can either return the 24-bit palette by casting the uint16_t pointer or alias another pointer to the same buffer. The only down side is that it will require 512 more bytes of RAM to hold the main and alternate palettes in RGB24 format. I'll work on this later in the week. Thanks for the suggest 👍

embedded-creations, 2020-11-04:
Glad you have a way to do it easily, I'm looking forward to trying it out when you're done!

bitbank2, 2020-11-06:
I pushed the change to the main branch. Please give it a try and if you're happy with it, I'll do a new release.
The change for you is that the begin() method has an optional second parameter to set the palette type (GIF_PALETTE_RGB888 or GIF_PALETTE_RGB565 / default). The GIFDRAW structure now has a uin8_t pointer for the 24-bit palette (which points to the same internal buffer).

embedded-creations, 2020-11-06:
Awesome!  I got it working, after modifying the `play_all_sd_files` example to run with SmartMatrix Library on the Teensy 4.1.

I don't think there's a rush to do the release until there's a working example, which I can contribute if that helps.  It may take me at least a few days though, as I want to fully understand the capabilities of your library and not just rush out a proof of concept example.

No big deal to me, but I see your change on the linux branch, not main.

bitbank2, 2020-11-06:
Thanks for testing it - oh - you're right about the branch. I'll fix that.

closed 2020-11-07 by bitbank2