bitbank2/AnimatedGIF #16: Add Error Codes for Debugging?

embedded-creations, 2020-11-07:
I just tried playing a handful of GIFs that work just fine with the [GifDecoder library](https://github.com/pixelmatix/gifdecoder) but only one of them opened with AnimatedGIF's play_all_sd_files sketch.  I'd like to debug but the library doesn't make it easy to get debug information out.  I'd like to add in return error codes for debugging as having people paste GIFs into GitHub Issues and ask "why doesn't this work?" doesn't scale very well.

e.g. https://github.com/pixelmatix/GifDecoder/blob/master/src/GifDecoder_Impl.h#L63

You've already defined that `GifInit()` returns 0 for an error, and I don't want to break existing sketches.  I'm thinking about adding error codes to `GIFParseInfo()`, and a new `GifInitDebug()` function that passes on the return codes from `GIFParseInfo()`.  Does that sound good, or do you have a better way to do it?

bitbank2, 2020-11-07:
Have you checked the return code from decode()? It may be because one or more of the frames uses the interlaced mode. I don't support it to not have to require the MCU to have lots of RAM. Please paste 1 failing image in this chat and I'll take a look.

embedded-creations, 2020-11-07:
Here's two GIFs that I attempted to debug.  I ran a bunch of GIFs through GifDecoder with debug flags turned on and the only pattern I found in the debug output is that all working GIFs had transparency enabled, and these two non-working had transparency disabled.  I found with AnimatedGIF these draw to the screen, but only black pixels.

[image]
[image]

bitbank2, 2020-11-07:
I just tested this on an ESP32 board and both GIF files you shared play perfectly.

https://youtu.be/HplufrDrYNY

It sounds like your drawing code is doing something wrong.

closed 2020-11-07 by bitbank2

embedded-creations, 2020-11-07:
Sure enough, it was a bug in my drawing code which only showed up in GIFs without transparency, sorry to waste your time!