bitbank2/AnimatedGIF #22: Handling of GIF_EMPTY_FRAME error is confusing for user

embedded-creations, 2021-02-07:
If someone calls `AnimatedGIF::playFrame(false, &delayMilliseconds)`, and gets a return code of 0 (good result and no more frames exist), they have to assume that a frame was processed and that they should delay for `delayMilliseconds`.  On GIFs that have an empty frame at the end, that results in the last frame being displayed for an extra unknown duration (could be negative!), as there was no frame processed and delayMilliseconds is unchanged.  (In my case `delayMilliseconds` is holding the same value as the previous call to `playFrame()` so the delay is twice as long as it should be).

https://github.com/bitbank2/AnimatedGIF/blob/master/src/AnimatedGIF.cpp#L196

I'm fixing this in the GifDecoder wrapper by loading a known bad value (-1) into `delayMilliseconds` and checking for that value if I get a return code of 0, indicating that there was a `GIF_EMPTY_FRAME` error, as I don't see any other way of determining that without modifying AnimatedGIF.

Suggested fix:

- Set `delayMilliseconds` to zero when seeing the `GIF_EMPTY_FRAME` error, so if someone uses `delayMilliseconds` they're not delaying for an unknown amount of time.
  - That fix is no good for me on it's own, as with GifDecoder I call an `updateScreenCallback` that should only be called if there's new data in the frame, and I also keep track of the number of frames in the GIF (incrementing after each successful call to `playFrame()`)
- Also keep the `GIF_EMPTY_FRAME` error, and don't replace with `GIF_SUCCESS`.
  - Most users won't be looking for the last error on `playFrame()` return of 0, and if they are, they can look for  both `GIF_EMPTY_FRAME` and `GIF_SUCCESS`.
  - I can use the combination of the return code and `getLastError()` to determine if there was a frame processed

bitbank2, 2021-02-07:
I just pushed a change which adds support for interlaced frames and addresses the issue you raise here. Please give it a try and if you're satisfied, I'll do a new release.

embedded-creations, 2021-02-07:
@bitbank2 that seems to fix the issue where `delayMilliseconds` is left with unset, but actually makes it worse for me wanting to know if a frame was processed.  Can you also keep the `GIF_EMPTY_FRAME` error, and don't replace with `GIF_SUCCESS`, or do you have another suggestion on how to know if a frame was processed after calling `playFrame()`?

bitbank2, 2021-02-07:
ok, please try again. I believe I have patched the last hole in the logic.

embedded-creations, 2021-02-07:
@bitbank2 The issue is still here in `playFrame()` (not in the C interface which I'm not using):

https://github.com/bitbank2/AnimatedGIF/blob/master/src/AnimatedGIF.cpp#L196

`GIF_EMPTY_FRAME` gets changed to `GIF_SUCCESS`, so it looks like a frame was processed, but it wasn't and there's no way for me to know the difference.

bitbank2, 2021-02-07:
You're right - I was working with the C code. I removed the offending line. If that's not it, please make the change and do a pull request.

embedded-creations, 2021-02-07:
Great!  I'll test this and an interlaced GIF as well and let you know

embedded-creations, 2021-02-07:
Needed one more line of code, and I also added some comments so just did a PR.  Feel free to copy/paste if that's easier than accepting the PR #23 

I tested the interlaced GIFs @PaintYourDragon posted in #14 and they work.  I didn't really look at the C code changes nor try to test them, LMK if you want me to specifically look at those.

bitbank2, 2021-02-07:
Thanks for the changes - accepted.

closed 2021-02-07 by bitbank2