bitbank2/AnimatedGIF #24: Hangs on gif.playFrame(true, NULL)

IonImpulse, 2021-02-11:
I've been trying to figure this out for a bit now, but after everything gets set up and I try to display the first frame, it hangs, not even giving an error code. I'm working on a PyPortal Titano, with everything up to that point working perfectly. Here's the code in the main loop:

```cpp
Serial.println("Starting Loop!");
    // read diagnostics (optional but can help debug problems)
    uint8_t x = tft.readcommand8(HX8357_RDPOWMODE);
    Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
    x = tft.readcommand8(HX8357_RDMADCTL);
    Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
    x = tft.readcommand8(HX8357_RDCOLMOD);
    Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
    x = tft.readcommand8(HX8357_RDDIM);
    Serial.print("Image Format: 0x"); Serial.println(x, HEX);
    x = tft.readcommand8(HX8357_RDDSDR);
    Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX); 

	Serial.println("About to call gif.open");
	if (gif.open("/lofi.gif", GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile, GIFDraw))
	{
		GIFINFO gi;
		Serial.printf("Successfully opened GIF; Canvas size = %d x %d\n", gif.getCanvasWidth(), gif.getCanvasHeight());
		if (gif.getInfo(&gi)) {
		Serial.printf("frame count: %d\n", gi.iFrameCount);
		Serial.printf("duration: %d ms\n", gi.iDuration);
		Serial.printf("max delay: %d ms\n", gi.iMaxDelay);
		Serial.printf("min delay: %d ms\n", gi.iMinDelay);
		Serial.println("4");
		}

		while (true)
		{
			Serial.println("5");
			Serial.printf("code: %d \n", gif.playFrame(true, NULL));
			delay(100);
		}
		gif.close();
	}
	else
	{
		Serial.printf("Error opening file = %d\n", gif.getLastError());
		while (1)
		{};
	}
```
and the output:

```cpp
Starting Loop!
Display Power Mode: 0x0
MADCTL Mode: 0x9C      
Pixel Format: 0x60
Image Format: 0x5
Self Diagnostic: 0x0
About to call gif.open
Successfully opened GIF; Canvas size = 480 x 320
frame count: 188
duration: 18800 ms
max delay: 100 ms
min delay: 100 ms
4
5
```

The 4 and the 5 are just there for simple debugging, to determine that it has reached that point. I've modified the MAX_WIDTH in AnimatedGif.h to equal 480, and the gif in question is 480x320. I've attached the file in question below:
[image]

Any idea why it would hang? Again, there's no result from the function, just infinite delay.

IonImpulse, 2021-02-11:
Ok I resized the gif down to 320x240, and now it's just random RGB noise. I'm beginning to think this might be a hardware problem, although I'm not 100% sure

bitbank2, 2021-02-11:
I merged a PR from a user without double checking his logic. The framedelay variable in playFrame() can be NULL, but the change he made wrote to the pointer anyway. Do a git pull on this code and see if it fixes it. Or you can just pass a pointer to an integer for the second parameter of playFrame().

IonImpulse, 2021-02-11:
Thank you so muc for the help! Unfortunately, this did not fix my problem. I've also noticed that my board hangs on tft.fillScreen(), but not on other gfx functions, such as tft.drawCircle(), so it might be some obscure hardware or firmware problem. Thanks again though!

bitbank2, 2021-02-12:
Any update on this issue? Can I close it?

IonImpulse, 2021-02-15:
Oh sorry, didn't see this until now. Yeah, I'm pretty sure it's my compiler of PIO on VSCode now, so I'll just close it. I'll send an update if I do link it back to this library though.

closed 2021-02-15 by IonImpulse