bitbank2/AnimatedGIF #26: Compilation error Example: play_all_sd_files.ino

tistructor, 2021-04-01:
Hello,
I used the Explora display with st7735r with integrated sd-card reader.
The core is ESP32 and the big_mem_demo example works fine. But when I insert SD-card this schetch has random pixels on the edges of the image even though the gif is spinning.
I kept HIGH sd_cs with a pull-up but to no avail.

Instead the example file play_all_sd_files.ino gives errors during the compilation due to problems of type concerning the callbacks.
I fixed the problems and now the file is compiled but the gif files on the sd-card are not displayed and they are opened and read correctly at least so it reports the log from serial.

```
#include <AnimatedGIF.h>
#include <bb_spi_lcd.h>
#include <SPI.h>
#include <SD.h>

// Demo sketch to play all GIF files in a directory
// Tested on TTGO T-Camera Plus

#define LED_PIN   16
#define DC_PIN    22
#define RES_PIN   21
#define CS_PIN    5
#define SCK_PIN   18
#define MISO_PIN  19
#define MOSI_PIN  23
#define SD_CS_PIN 17

#define DISPLAY_WIDTH 128
#define DISPLAY_HEIGHT 160

SPILCD lcd;

AnimatedGIF gif;
File f;
int x_offset, y_offset;

// Draw a line of image directly on the LCD
void GIFDraw(GIFDRAW *pDraw)
{
    uint8_t *s;
    uint16_t *d, *usPalette, usTemp[320];
    int x, y, iWidth;

    iWidth = pDraw->iWidth;
    if (iWidth > DISPLAY_WIDTH)
       iWidth = DISPLAY_WIDTH;
    usPalette = pDraw->pPalette;
    y = pDraw->iY + pDraw->y; // current line
    
    s = pDraw->pPixels;
    if (pDraw->ucDisposalMethod == 2) // restore to background color
    {
      for (x=0; x<iWidth; x++)
      {
        if (s[x] == pDraw->ucTransparent)
           s[x] = pDraw->ucBackground;
      }
      pDraw->ucHasTransparency = 0;
    }
    // Apply the new pixels to the main image
    if (pDraw->ucHasTransparency) // if transparency used
    {
      uint8_t *pEnd, c, ucTransparent = pDraw->ucTransparent;
      int x, iCount;
      pEnd = s + iWidth;
      x = 0;
      iCount = 0; // count non-transparent pixels
      while(x < iWidth)
      {
        c = ucTransparent-1;
        d = usTemp;
        while (c != ucTransparent && s < pEnd)
        {
          c = *s++;
          if (c == ucTransparent) // done, stop
          {
            s--; // back up to treat it like transparent
          }
          else // opaque
          {
             *d++ = usPalette[c];
             iCount++;
          }
        } // while looking for opaque pixels
        if (iCount) // any opaque pixels?
        {
          spilcdSetPosition(&lcd, pDraw->iX+x+x_offset, y+y_offset, iCount, 1, 1);
          spilcdWriteDataBlock(&lcd, (uint8_t *)usTemp, iCount*2, 1);
          x += iCount;
          iCount = 0;
        }
        // no, look for a run of transparent pixels
        c = ucTransparent;
        while (c == ucTransparent && s < pEnd)
        {
          c = *s++;
          if (c == ucTransparent)
             iCount++;
          else
             s--; 
        }
        if (iCount)
        {
          x += iCount; // skip these
          iCount = 0;
        }
      }
    }
    else
    {
      s = pDraw->pPixels;
      // Translate the 8-bit pixels through the RGB565 palette (already byte reversed)
      for (x=0; x<iWidth; x++)
        usTemp[x] = usPalette[*s++];
      spilcdSetPosition(&lcd, pDraw->iX+x_offset, y+y_offset, iWidth, 1, 1);
      spilcdWriteDataBlock(&lcd, (uint8_t *)usTemp, iWidth*2, 1);
    }
} /* GIFDraw() */

void * GIFOpenFile(const char *fname, int32_t *pSize)
{
  f = SD.open(fname);
  if (f)
  {
    *pSize = f.size();
    return (void *)&f;
  }
  return NULL;
} /* GIFOpenFile() */

void GIFCloseFile(void *pHandle)
{
  File *f = static_cast<File *>(pHandle);
  if (f != NULL)
     f->close();
} /* GIFCloseFile() */

int32_t GIFReadFile(GIFFILE *pFile, uint8_t *pBuf, int32_t iLen)
{
    int32_t iBytesRead;
    iBytesRead = iLen;
    File *f = static_cast<File *>(pFile->fHandle);
    // Note: If you read a file all the way to the last byte, seek() stops working
    if ((pFile->iSize - pFile->iPos) < iLen)
       iBytesRead = pFile->iSize - pFile->iPos - 1; // <-- ugly work-around
    if (iBytesRead <= 0)
       return 0;
    iBytesRead = (int32_t)f->read(pBuf, iBytesRead);
    pFile->iPos = f->position();
    return iBytesRead;
} /* GIFReadFile() */

int32_t GIFSeekFile(GIFFILE *pFile, int32_t iPosition)
{ 
  int i = micros();
  File *f = static_cast<File *>(pFile->fHandle);
  f->seek(iPosition);
  pFile->iPos = (int32_t)f->position();
  i = micros() - i;
//  Serial.printf("Seek time = %d us\n", i);
  return pFile->iPos;
} /* GIFSeekFile() */

void setup() {
  Serial.begin(115200);
  while (!Serial);

// Need to initialize SPI before calling SD.begin()
  SPI.begin(SCK_PIN, MISO_PIN, MOSI_PIN, SD_CS_PIN);
  if (!SD.begin(SD_CS_PIN)) {
    Serial.println("SD card init failed!");
    while (1); // SD initialisation failed so wait here
  }
  Serial.println("SD Card init success!");

  spilcdInit(&lcd, LCD_ST7735R, FLAGS_NONE, 32000000, CS_PIN, DC_PIN, RES_PIN, LED_PIN, MISO_PIN, MOSI_PIN, SCK_PIN);
  gif.begin(BIG_ENDIAN_PIXELS);
}

void ShowGIF(const char *name)
{
  spilcdFill(&lcd,1,1);
  
  if (gif.open(name, GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile, GIFDraw))
  {
    x_offset = (DISPLAY_WIDTH - gif.getCanvasWidth())/2;
    if (x_offset < 0) x_offset = 0;
    y_offset = (DISPLAY_HEIGHT - gif.getCanvasHeight())/2;
    if (y_offset < 0) y_offset = 0;
    Serial.printf("Successfully opened GIF; Canvas size = %d x %d\n", gif.getCanvasWidth(), gif.getCanvasHeight());
    Serial.flush();
    while (gif.playFrame(true, NULL))
    {      
    }
    gif.close();
  }

} /* ShowGIF() */

//
// Return true if a file's leaf name starts with a "." (it's been erased)
//
int ErasedFile(char *fname)
{
int iLen = strlen(fname);
int i;
  for (i=iLen-1; i>0; i--) // find start of leaf name
  {
    if (fname[i] == '/')
       break;
  }
  return (fname[i+1] == '.'); // found a dot?
}
void loop() {

char *szDir = "/GIF"; // play all GIFs in this directory on the SD card
char fname[256];
File root, temp;

   while (1) // run forever
   {
      root = SD.open(szDir);
      if (root)
      {
         temp = root.openNextFile();
            while (temp)
            {
              if (!temp.isDirectory()) // play it
              {
                strcpy(fname, temp.name());
                if (!ErasedFile(fname))
                {
                  Serial.printf("Playing %s\n", temp.name());
                  Serial.flush();
                  ShowGIF((char *)temp.name());
                }
              }
              temp.close();
              temp = root.openNextFile();
            }
         root.close();
      } // root
      delay(4000); // pause before restarting
   } // while
} /* loop() */
```
Serial Terminal
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
SD Card init success!
Playing /GIF/webview.gif
Successfully opened GIF; Canvas size = 128 x 160
Playing /GIF/contacts.gif
Successfully opened GIF; Canvas size = 128 x 128
Playing /GIF/weather.gif
Successfully opened GIF; Canvas size = 128 x 160
Playing /GIF/webview.gif
Successfully opened GIF; Canvas size = 128 x 160
Playing /GIF/contacts.gif
Successfully opened GIF; Canvas size = 128 x 128
Playing /GIF/weather.gif
Successfully opened GIF; Canvas size = 128 x 160
Playing /GIF/webview.gif
Successfully opened GIF; Canvas size = 128 x 160
Playing /GIF/contacts.gif
Successfully opened GIF; Canvas size = 128 x 128
Playing /GIF/weather.gif
Successfully opened GIF; Canvas size = 128 x 160
Playing /GIF/webview.gif
Successfully opened GIF; Canvas size = 128 x 160
```
I forgot, there is a compilation error in big_mem_demo too
this buffer is declared in the ucTXBuf example file, but it is also declared in the library, it was enough to change the name slightly and the problem was solved.

I am using the official library that is downloaded inside arduino 1.8.13

tistructor, 2021-04-01:
Furthermore, to remove the random pixels on the edges of the screen it is not enough to remove the sd-card from the slot or power supply to ESP32 and display but it is necessary to load a skecth with an example of adafruit and then reload the example big_mem_demo in this way the display returns to normal.

tistructor, 2021-04-02:
With an arduino due card I was able to run the adafruit_gfx_sdcard example by loading a gif file from the sd-card of an external module. I have not used the one integrated in the explora display (sst7735R).
The gif is displayed but lines with random pixels always appear below and on the left edge of the screen, precisely: 2 lines on the left edge and 1 line below.
If I can post a photo.

[https://ibb.co/7YRFvKF](url)

tistructor, 2021-04-02:
Problem solved.
1.
There was an initialization problem in the adafruit file "Adafruit_ST7735.cpp" in setRotation function
```
switch (rotation) {
  case 0:
    if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
      madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;
    } else {
      madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB; // WRITE this line with RGB (first BGR)
    }
```
I used this initialization:
"tft.initR (INITR_GREENTAB); // Init ST7735S chip, green tab"

2.
Incorrect example code in the file: "play_all_sd_files.ino"

Wrong path to load gif file from sd-card
```
void loop () {

char * szDir = "/ GIF"; // play all GIFs in this directory on the SD card
char fname [256];
Root file, temp;

   while (1) // run forever
   {
      root = SD.open (szDir);
      if (root)
      {
         temp = root.openNextFile ();
            while (temp)
            {
              if (! temp.isDirectory ()) // play it
              {
/ ************ CHANGE ***************** /
                strcpy (fname, szDir);
                strcat (fname, "/");
                strcat (fname, temp.name ());
/ ************ ************************** /
                if (! ErasedFile (fname))
                {
                  Serial.print ("Playing"); Serial.println (temp.name ());
                  Serial.println (fname);
                  Serial.flush ();
                  ShowGIF ((char *) fname);
                }
              }
              temp.close ();
              temp = root.openNextFile ();
            }
         root.close ();
      } // root
      // delay (2000); // pause before restarting
   } // while
} / * loop () * /
```

tistructor, 2021-04-02:
The modification to determine the correct path only applies to arduino due.
For the ESP32 card it is necessary to restore the original algorithm.
The SD library is implemented differently between arduino due and esp32.

Thanks for the work done

jhonnyx161, 2021-04-09:
> Problema resuelto.
> 1.
> Hubo un problema de inicialización en el archivo adafruit "Adafruit_ST7735.cpp" en la función setRotation
> 
> ```
> switch (rotation) {
>   case 0:
>     if ((tabcolor == INITR_BLACKTAB) || (tabcolor == INITR_MINI160x80)) {
>       madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB;
>     } else {
>       madctl = ST77XX_MADCTL_MX | ST77XX_MADCTL_MY | ST77XX_MADCTL_RGB; // WRITE this line with RGB (first BGR)
>     }
> ```
> 
> 
> Usé esta inicialización: "tft.initR (INITR_GREENTAB); // Init ST7735S chip, tab verde"
> 
> 
> Código de ejemplo incorrecto en el archivo: "play_all_sd_files.ino"
> 
> Ruta incorrecta para cargar el archivo gif desde la tarjeta sd
> 
> ```
> void loop () {
> 
> char * szDir = "/ GIF"; // play all GIFs in this directory on the SD card
> char fname [256];
> Root file, temp;
> 
>    while (1) // run forever
>    {
>       root = SD.open (szDir);
>       if (root)
>       {
>          temp = root.openNextFile ();
>             while (temp)
>             {
>               if (! temp.isDirectory ()) // play it
>               {
> / ************ CHANGE ***************** /
>                 strcpy (fname, szDir);
>                 strcat (fname, "/");
>                 strcat (fname, temp.name ());
> / ************ ************************** /
>                 if (! ErasedFile (fname))
>                 {
>                   Serial.print ("Playing"); Serial.println (temp.name ());
>                   Serial.println (fname);
>                   Serial.flush ();
>                   ShowGIF ((char *) fname);
>                 }
>               }
>               temp.close ();
>               temp = root.openNextFile ();
>             }
>          root.close ();
>       } // root
>       // delay (2000); // pause before restarting
>    } // while
> } / * loop () * /
> ```

THANK YOU, YOUR CONTRIBUTION HELP TO SOLVE A PROBLEM WITH STM32 F4

bitbank2, 2021-04-21:
I pushed changes to update the 2 sketches to build/work correctly with the latest version of my bb_spi_lcd library. Please clone the new repo and see if they work for you, then I'll do a new release.

tistructor, 2021-04-21:
Hello,
I tried the following sample files:
adafruit_gfx_memory.ino
adafruit_gfx_sdcard.ino
big_mem_demo.ino
gif_benchmark.ino
play_all_sd_files.ino

With ESP32 and ILI9341 and they work by changing the pins with respect to my circuit.

The play_all_sd_files.ino is slow compared to the version I made using the libt¡reria adafruit and I don't understand why, it seems to go with SPI software.
I set the bus to 40MHz as in the adafruit library but the frame rate is low.

In the example file big_mem_demo.ino I am sure that spi is hardware since I tried to put 80MHz and the program runs but the image is corrupted.

bitbank2, 2021-04-21:
It may have to do with the use of DMA which needs to be explicitly requested with my library.

closed 2021-04-21 by bitbank2