bitbank2/AnimatedGIF #34: API Change in 1.4.0 breaks existing sketches

embedded-creations, 2021-06-04:
The change to `AnimatedGIF::begin()` removing `iEndian` is a large API change, and I believe along with deleting the definition for `LITTLE_ENDIAN_PIXELS`/`BIG_ENDIAN_PIXELS` broke a bunch of your example code. Maybe you should keep the endian definitions and the two-argument begin() calls around for backwards compatibility? I'm not sure how to support this change with my GifDecoder library. I'd probably have to note that earlier versions of my library are only compatible with AnimatedGIF <1.4.0 and later versions are only compatible with >=1.4.0

It's already causing problems for my GifDecoder Library users: https://github.com/pixelmatix/GifDecoder/issues/8

bitbank2, 2021-06-05:
You're right; I can add the new feature without breaking things. I'll take care of this tomorrow.

r-colvin, 2021-06-05:
@embedded-creations was just on my way to report the same! Cheers mate - @bitbank2  thanks for fixing this so quick, appreciate it mate

bitbank2, 2021-06-06:
I just pushed a fix; please clone and test, then I'll do a new release.

closed 2021-06-06 by bitbank2

embedded-creations, 2021-06-06:
That doesn't fully fix my issue, but adding this one line to AnimatedGIF.h does:

```
    void begin(int iEndian, unsigned char ucPaletteType) { begin(ucPaletteType); }; // Only needed for backwards compatibility
```

Unlike all your examples, I'm using `GIF_PALETTE_RGB888` so my code uses both arguments of `begin()`.

https://github.com/pixelmatix/GifDecoder/blob/master/src/GifDecoder_Impl.h#L269

Without the above backwards compatibility fix I'd have to note that earlier versions of my library are only compatible with AnimatedGIF <1.4.0 and later versions are only compatible with >=1.4.0

bitbank2, 2021-06-06:
ok, added it. Let me know if it works again.

embedded-creations, 2021-06-06:
Working, thanks!

bitbank2, 2021-06-06:
Great - I'll release 1.4.1
Thanks for your help