bitbank2/AnimatedGIF #36: PlatformIO library ref issues

Stumblor, 2021-06-08:
Looks as though the repo location has changed recently in PlatformIO?

Formally:
bitbank2/AnimatedGIF

Now:
laurencebank@gmail.com/AnimatedGIF

This is leading to problems:
[image]

bitbank2, 2021-06-08:
I'm sorry about that, but I created it for the Arduino library manager. I don't know how PlatformIO works nor how this happened. If there is something I can change from my side, let me know.

Stumblor, 2021-06-08:
No worries Gary - I've raised on the platformIO repo as well, but not sure how quickly they respond to that (the oldest issue of this type was 3 months ago!)

Just using a direct reference for the moment.

tobozo, 2021-06-08:
platformio doesn't use github "releases" channel, they produce the library packages directly from the main branch and only trust the contents of library.json/library.properties file for the version number.

A quick solution when this happens is to register an account on their forum, produce a token and issue `pio package`  `login` / `unpublish` / `publish` at the command prompt from said library folder

tobozo, 2022-08-11:
Here's a snippet to auto-publish this library to platformio registry when a new release is published.

It depends on a Github Action Secret `PLATFORMIO_AUTH_TOKEN` *(project settings/security/secret/action).

`.github/workflows/PlatformioPackagePublish.yml`

```yaml

name: PioPackagePublish

on:
  release:
    types: [published]
  workflow_dispatch:

jobs:
  build:
    runs-on: ubuntu-latest

    steps:
      - name: Checkout
        uses: actions/checkout@v2
        with:
          ref: ${{ github.event.pull_request.head.sha }}

      - name: Cache pip
        uses: actions/cache@v2
        with:
          path: ~/.cache/pip
          key: ${{ runner.os }}-pip-${{ hashFiles('**/requirements.txt') }}
          restore-keys: |
            ${{ runner.os }}-pip-
      - name: Cache PlatformIO
        uses: actions/cache@v2
        with:
          path: ~/.platformio
          key: ${{ runner.os }}-${{ hashFiles('**/lockfiles') }}
      - name: Set up Python
        uses: actions/setup-python@v2

      - name: Install PlatformIO
        run: |
          python -m pip install --upgrade pip
          pip install --upgrade platformio

      - name: Run PlatformIO
        env:
          PLATFORMIO_AUTH_TOKEN: ${{ secrets.PLATFORMIO_AUTH_TOKEN }}
        run: |
          pio package publish --owner bitbank2 --non-interactive

```

closed 2024-02-18 by bitbank2