bitbank2/AnimatedGIF #38: GIFs created with gifencoder.js have trailing zeros, and won't parse correctly

embedded-creations, 2021-06-18:
The issue I'm having is specifically with exports from noisedeck.app, which I found uses this for encoding GIFs under the hood: https://github.com/jnordberg/gif.js

Three example GIFs:

[image]

You can only change the GIF dimensions and number of frames in the pro version
[image]

Yes this is ugly but maybe it's useful to have a GIF with just two frames for brevity
[image]

There's trailing zeros left after the end of the GIF:

[image]

I'm guessing this is a result of the encoder using pages for storage and not trimming the excess at the end.

https://github.com/jnordberg/gif.js/blob/master/src/GIFEncoder.js#L19

`GIFParseInfo()` fails here:

https://github.com/bitbank2/AnimatedGIF/blob/master/src/gif.inl#L401

Here's my quick fix.  There's probably a better place to look for the trailer byte, but this works:

```
        else // invalid byte, stop decoding
        {
            if (pPage->GIFFile.iSize - iStartPos < 32) {
                // non-image bytes at end of file?
                pPage->iError = GIF_EMPTY_FRAME;

            } else if(p[iOffset] == ';') {
                // Trailer Byte - End of the GIF Data Stream
                pPage->iError = GIF_EMPTY_FRAME;
            }
            else {
                /* Bad header info */
                pPage->iError = GIF_DECODE_ERROR;
                Serial.print("byte: ");
                Serial.print(p[iOffset], HEX);
                Serial.print(" pos: ");
                Serial.print(iOffset, HEX);
            }
            return 0;
        }
```

bitbank2, 2021-06-23:
I did a cleaner fix which correctly handles your test files. Please try my last checkin and let me know if it works for you.

bitbank2, 2021-06-25:
Open a new issue if you find any other gifs which don't decode correctly.

closed 2021-06-25 by bitbank2

embedded-creations, 2021-07-06:
Sorry I didn't have a chance to try until now, the GIFs I was having trouble with work well now, thanks!