bitbank2/AnimatedGIF #42: Huge ram usage

tsctrl, 2021-06-27:
Hi Larry, 

thanks for this great library and sorry for spamming your repo.
after i enabled the AnimatedGif it consume around 23,988 bytes of RAM which quite huge.

is there option/params/config in the library that i can tweak or change to minimize the RAM usage as per my use / needed?
do this based on the screen width and height calculation?

thanks,
ts

bitbank2, 2021-06-27:
Hi ts,
I designed the library to not use malloc/free so that you could manage the memory. The AnimatedGIF class/structure has 23k of buffers to do the necessary work. Running the code won't consume any additional memory. You can instantiate the class statically or dynamically or place it in TCM/PSRAM/DDR as you please. You can potentially shrink it a little by reducing the size of the buffers in the class, but you will probably break it in the process. GIF decoding takes lots of memory to hold the dictionary, current line, palette(s), and file buffer.

closed 2021-06-27 by bitbank2

tsctrl, 2021-07-06:
hi @bitbank2 ,

could share the approach how to instantiate the class dynamically? does it mean it still consume 23k of memory after instantiated? not sure the purpose there.

thanks!

bitbank2, 2021-07-06:
Statically:
AnimatedGIF gif; // uses 23k of RAM

Dynamically:
AnimatedGIF *pGIF;
pGIF = new AnimatedGIF; // uses the same 23K of RAM, but this time from the heap