bitbank2/AnimatedGIF #44: Loading Gif into memory from Web

robrichardson13, 2021-07-21:
Hi @bitbank2 thanks for providing us with an awesome library! Do you have any suggestions/examples on how I can pull down a GIF hosted from a server into memory and then pass that into the library to `open()` it?

bitbank2, 2021-07-21:
What you're asking to do is pretty simple. What system do you need this to run on? I assume you're talking about ESP32. There are plenty of examples for ESP32 of downloading a file from a web server and once you have it in memory, just pass it to my library to decode. I'll ask the standard questions - what have you tried and what results did you get? The example sketches in this repo show you the code needed to decode an image from memory.

closed 2021-07-21 by bitbank2

robrichardson13, 2021-07-21:
Thanks for getting back to me.

I ended up getting this to work by having the server return back a base_64 encoded gif and then using this decode library to to decode it into a C binary to then pass to the AnimatedGIF library.
https://github.com/Densaugeo/base64_arduino