bitbank2/AnimatedGIF #46: Questions concerning rendering from SPIFFS.

avdz, 2021-12-09:
Hi,

I successfully rendered a gif (converted to .h file) from memory on my TTGO T-Dispaly (ESP32 with 16MB flash and a TFT_eSPI display).
I wanted to try render the same gif from flash memory. I successfully saved the .gif in my filesystem and can open the file, however I don't see anything appearing on the screen.+
Can I use the same GIFDRAW callback as the TFT_eSPI_memory example, while using the  GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile functions from the SD examples? (with GIF open modified to use SPIFFS).

Btw the gif I'm trying to use is this one:

[image]

bitbank2, 2021-12-09:
You need to ensure that the open/read/seek/close callbacks correctly access the data whether it's coming from SPIFFS, SD card or some other place. If the callback functions are doing their job correctly, the decoding will succeed. The GIFDraw function shouldn't need to change if you're using the same display.

closed 2021-12-09 by bitbank2

avdz, 2021-12-09:
Thank you for the answer. I'll look into the callback functions. 
I understand what the open/read & close functions should do. However I'm not clear about the function of the seek function. Do you mind clarifying this a little bit?

bitbank2, 2021-12-09:
Read normally advances the current read position by the number of bytes requested. If you want to read data that's before or after that point in the file, you need to use the Seek function to set the file pointer to a different spot. See:

https://man7.org/linux/man-pages/man3/fseek.3.html