bitbank2/AnimatedGIF #47: Example TFT_eSPI_memory crash on ESP8266

ifengchao, 2021-12-16:
Hi, nice work first ! 

I think it crashed here in git.playFrame(true,NULL);

Changes i made:
gif.begin(LITTLE_ENDIAN_PIXELS);// little endian for esp8266, board esp-12e
void GIFDraw(GIFDRAW *pDraw) copyed from GIFDRAW.ino
#define GIF_IMAGE ucBadgers   //  No DMA  63 fps, DMA:  71fps // tested this gif image

TFT Screen is st7789 240x240 size.

Logs printed on serial console:
Successfully opened GIF; Canvas size = 160 x 120
                                                 tft size = 240 x 240

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
...

Any help will be appreciated, thanks.

bitbank2, 2021-12-16:
I need a bit more detail to see why it's crashing. I can tell you already that the 8266 is troublesome because of the way it uses RAM to execute code and can have an acute lack of RAM at inopportune times. If you could, I would try the same code on something with more RAM, like the ESP32. If the ESP8266 is all you've got, then try by a process of elimination such as having an empty GIFDraw() callback as your first test. If that works, then you know it's in your code that the problem lies.

ifengchao, 2021-12-17:
> Thanks for the quick reply. Here is the main.cpp in my PlatformIO project.

// TFT_eSPI_memory
//
// Example sketch which shows how to display an
// animated GIF image stored in FLASH memory
//
// written by Larry Bank
// bitbank@pobox.com
//
// Adapted by Bodmer for the TFT_eSPI Arduino library:
// https://github.com/Bodmer/TFT_eSPI
//
// To display a GIF from memory, a single callback function
// must be provided - GIFDRAW
// This function is called after each scan line is decoded
// and is passed the 8-bit pixels, RGB565 palette and info
// about how and where to display the line. The palette entries
// can be in little-endian or big-endian order; this is specified
// in the begin() method.
//
// The AnimatedGIF class doesn't allocate or free any memory, but the
// instance data occupies about 22.5K of RAM.

//#define USE_DMA       // ESP32 ~1.25x single frame rendering performance boost for badgers.h
                        // Note: Do not use SPI DMA if reading GIF images from SPI SD card on same bus as TFT
#define NORMAL_SPEED  // Comment out for rame rate for render speed test

// Load GIF library
#include <AnimatedGIF.h>
AnimatedGIF gif;

// Example AnimatedGIF library images
#include "../test_images/badgers.h"
//#include "../test_images/homer.h"
//#include "../test_images/homer_tiny.h"
//#include "../test_images/pattern.h"

// ESP32 40MHz SPI single frame rendering performance
// Note: no DMA performance gain on smaller images or transparent pixel GIFs
  #define GIF_IMAGE ucBadgers   //  No DMA  63 fps, DMA:  71fps
//#define GIF_IMAGE ucHomer     //  No DMA 162 fps, DMA: 141 fps
//#define GIF_IMAGE homer_tiny  //  No DMA 564 fps, DMA: 481 fps
//#define GIF_IMAGE ucPattern   //  No DMA  90 fps, DMA:  78 fps

#include <SPI.h>
#include <TFT_eSPI.h>

TFT_eSPI tft = TFT_eSPI();

void setup() {
  Serial.begin(115200);

  tft.begin();
#ifdef USE_DMA
  tft.initDMA();
#endif
  tft.setRotation(1);
  tft.fillScreen(TFT_BLACK);

  gif.begin(LITTLE_ENDIAN_PIXELS);
}

// GIFDraw is called by AnimatedGIF library frame to screen

#define DISPLAY_WIDTH  tft.width()
#define DISPLAY_HEIGHT tft.height()
#define BUFFER_SIZE 64            // Optimum is >= GIF width or integral division of width

#ifdef USE_DMA
[... 140 lines cut ...]
#else // Test maximum rendering speed
void loop()
{
  long lTime = micros();
  int iFrames = 0;

  if (gif.open((uint8_t *)GIF_IMAGE, sizeof(GIF_IMAGE), GIFDraw))
  {
    tft.startWrite(); // For DMA the TFT chip slect is locked low
    while (gif.playFrame(false, NULL))
    {
      // Each loop renders one frame
      iFrames++;
      yield();
    }
    gif.close();
    tft.endWrite(); // Release TFT chip select for other SPI devices
    lTime = micros() - lTime;
    Serial.print(iFrames / (lTime / 1000000.0));
    Serial.println(" fps");
  }
}
#endif

Empty GIFDraw() will still crash, but if i comment "while (gif.playFrame(false, NULL))" in loop()，crash stopped.

bitbank2, 2021-12-17:
I don't use platformio, so I don't know if that's affecting things, but I'm still not seeing anything that could help me diagnose the problem. If you can zip up the whole project including the GIF files you're using, I can take a look this weekend. Email it to bitbank@pobox.com

ifengchao, 2021-12-17:
> I don't use platformio, so I don't know if that's affecting things, but I'm still not seeing anything that could help me diagnose the problem. If you can zip up the whole project including the GIF files you're using, I can take a look this weekend. Email it to [bitbank@pobox.com](mailto:bitbank@pobox.com)

Actually main.cpp could be renamed to main.ino (almost the same with your example code in TFT_eSPI_memory.ino) and open in Arduino IDE, the gif image i'm using is defined in #include "../test_images/badgers.h". I tried in Arduino, but also crashed. Do you happen to have arduino enviroment and an esp8266 board, it's easy to reproduce the crash.

ifengchao, 2021-12-17:
I hope this will not take you too much time,  if it could not be reproduced in your enviroment, i may move to esp32 few days later,  need to do some screen soldering work. :)

ifengchao, 2021-12-17:
> I don't use platformio, so I don't know if that's affecting things, but I'm still not seeing anything that could help me diagnose the problem. If you can zip up the whole project including the GIF [...]

Another thing i forgot to declear is i made some changes to User_Setup.h in TFT_eSPI library to make the st7789 spi screen work, and it worked with test code without AnimatedGif library. So i think this part may not be the reason caused crash.

bitbank2, 2021-12-17:
I do have an 8266 in my collection of parts; I'll give it a try later today.

bitbank2, 2021-12-17:
Found the issue... ESP8266 is a Harvard architecture and my generic open function treats the source memory like RAM. In my JPEG decoder I added a specific function to deal with FLASH reads from a different address space. A quick fix for you is to edit gif.inl and change the memcpy() in readMem() to be memcpy_P(). I'll have to add a new function called "openFLASH()" to deal with the 8266 specifically.

ifengchao, 2021-12-18:
> 

Crash fix confirmed, now it can display small smooth gif animations. Great job, i'll do more test. Thanks Larry
[image]
.

closed 2021-12-18 by ifengchao

bitbank2, 2021-12-18:
I did a new release (1.4.5) which adds the openFLASH method