bitbank2/AnimatedGIF #48: Image edges are jagged on esp8266

ifengchao, 2021-12-19:
Hi Larry, i was able to play gifs stored with littlefs in nor flash.
But i found most gif images edges are jagged, especially image edges with small lines.
https://youtu.be/CwPEhcc7Wx0

I've tried diffrent TFT_eSPI settings, but can't help.
I wonder if it's limits on esp8266, or i need do some settings with AnimatedGIF library, or something limited by the sreen?

I've tested jpg and bmp images display with no such obvious problem.

Board: esp8266
Screen: st7789 spi screen
Tested gif image:
[image]
 
Thanks Larry.

bitbank2, 2021-12-19:
Have a look at this video I took of a ST7789 240x240 LCD playing your GIF file. The animation looks relatively smooth, but there are limitations due to the color palette. SPI LCDs are typically 16-bits per pixel (RGB565). Depending on the color choices, 5 bits of red and blue can look grainy and edges which use anti-aliasing can look jagged. This is not a problem of the software, but of reducing a RGB888 palette to RGB565.

https://youtu.be/b19EahUKJk8

closed 2021-12-19 by bitbank2

ifengchao, 2021-12-20:
> 

Your video looks quite good!
Finally found that i misunderstood the ENDIAN settings in gif.begin() with the CPU's endian(esp8266 is little endian).
Now i started with gif.begin(GIF_PALETTE_RGB565_BE), all works great !!!
Thanks again Larry.

bitbank2, 2021-12-20:
Yes, using the wrong endianness would make the pixels look quite strange :)