bitbank2/AnimatedGIF #49: Can I use this for just decoding GIF images?

sam20908, 2021-12-23:
I'm currently using `stbi__load_gif_main` to load GIF files, but it is too slow for me (because of dynamic allocation). This library seemed interesting, as it states that it doesn't use dynamic allocation, but I want to use it for desktop applications, not Arduino.

If this library isn't applicable to my use case, could you recommend me a fast GIF decoder library for C++, or is dynamic allocation inevitable?

Thanks!

bitbank2, 2021-12-23:
You can certainly use my library for GIF images on a Desktop, and it's faster than the "standard" GIF implementation. It's true that it doesn't do any dynamic allocation. You can either give it a buffer for the entire output image or have it call your callback function a line at a time. There's an example in the /linux folder that should build on any POSIX OS.

closed 2021-12-23 by bitbank2

sam20908, 2021-12-23:
> You can either give it a buffer for the entire output image or have it call your callback function a line at a time.
Can you explain what you mean by a line at a time? My idea was to somehow allow different images to be in different buffers, so I can pass the buffer to other libraries.
Also, could you provide a code example to achieve the above?

> There's an example in the /linux folder that should build on any POSIX OS.
I am using Windows, does it work on that?

bitbank2, 2021-12-24:
Take a look in the Wiki for details about everything.
I don't use Windows, but I assume with the "Linux cli" support it could compile without issue.

sam20908, 2021-12-24:
Sorry to keep bothering you, I've looked at the wiki and searched elsewhere, but with the `GIF_DRAW_CALLBACK` giving me a `GIFDRAW*`, I don't fully understand how to take the `pPixels` member and putting it in a buffer that I can eventually use (because I don't work with LCD). Initially, I thought that I can just fill up my buffer line by line by memcpy'ing `pPixels` by the length of `iWidth * 4` (because of RGBA for each pixel, and because it's a line). I couldn't find anything that confirms my guess, and the LCD code didn't really help either.

bitbank2, 2021-12-24:
After each scanline is decoded, the GIF_DRAW_CALLBACK is called with the row of raw 8-bit pixels and the palette info. You can convert the pixels into any form you like. You're given the line number of the current frame along with its x/y offset within the image (each frame can be a sub-region). GIF pixels contain 1-8 bits of palette color info each (not RGBA8888). It's up to you to convert them to another form if you want.