bitbank2/AnimatedGIF #50: Play SD card gif with tft screen on esp8266

ifengchao, 2021-12-25:
Hello Larry, sorry to bother you again, but I have tried all the methods I thought of and there is no better idea. Have to find an expert like you to give some clues. I tried to share SPI with TFT screen and SD card on ESP8266 platform, after debugging for a lot of time, finally it worked, but the effect is not as expected. My original purpose was to transcode the video to gif format through ffmpeg on the Linux platform, and then stored on the SD card of the ESP8266 for smooth playback. The gif image used in the test was about 50MB in size. The problem I am facing now one is that it freezes every frame when playing, and the other is that the program crashes after playing for a while. I am wondering whether this is an impossible goal on ESP8266. If you have free time, and at the same time if it is convenient for you, I can send you the code and image. I want to figure out why it can't be achieved, maybe finally have to switch to ESP32 after the last try.

The following video is the problem I encountered:
https://www.youtube.com/watch?v=pcBab8AUj-Q

Thank you again for your time.

bitbank2, 2021-12-25:
Share the code and I'll take a look.

ifengchao, 2021-12-25:
> 

OK, please check your email box.

ifengchao, 2021-12-26:
Move to ESP32 finally. But play gif stored in littlefs with esp8266 without problem.This library is excellent!

closed 2021-12-26 by ifengchao

bitbank2, 2021-12-26:
Great - glad you got it working