bitbank2/AnimatedGIF #52: gif larger than 320 crash

Koxx3, 2022-02-20:
Hello,
i am trying to play gif larger than 320 on ESP32.
I modified MAX_WIDTH in .h file, but it crashs.
It seems to come from DecodeLZW or GIFMakePels.
Any idea what could fail ?

stack trace doesn't git much informations ...

```
09:44:47.632 > Guru Meditation Error: Core  1 panic'ed (Double exception).
09:44:47.633 >
09:44:47.633 > Core  1 register dump:
09:44:47.633 > PC      : 0x40092223  PS      : 0x00040636  A0      : 0x800e5b22  A1      : 0x3ffecf70
09:44:47.634 > A2      : 0x3ffed220  A3      : 0x3fff26a6  A4      : 0x000000ff  A5      : 0x00000000
09:44:47.635 > A6      : 0x3fff26a6  A7      : 0x3fff36b0  A8      : 0x0000015e  A9      : 0x00000000
09:44:47.636 > A10     : 0x00000000  A11     : 0x00000000  A12     : 0x0000015e  A13     : 0x00000001
09:44:47.637 > A14     : 0x3ffecf70  A15     : 0xff000000  SAR     : 0x00000015  EXCCAUSE: 0x00000002  
09:44:47.638 > EXCVADDR: 0xfffffff1  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
09:44:47.639 >
09:44:47.639 >
09:44:47.639 > Backtrace:0x40092220:0x3ffecf700x400e5b1f:0x3ffed220 0x400e5b1f:0x00000000  |<-CORRUPTED
09:44:47.687 >   #0  0x40092220:0x3ffecf700 in _xt_context_save at C:\Users\Francois\.platformio\packages\framework-espidf\components\freertos\port\xtensa/xtensa_context.S:194
09:44:47.687 >
09:44:47.687 >
09:44:47.687 >
09:44:47.687 >
```

bitbank2, 2022-02-20:
You haven't provided enough info to know where the problem might be. Did you provide a frame buffer to draw into? Is your data read callback set up correctly?

Koxx3, 2022-02-20:
no, I think it's fine ;)
I just used an old example of something like than which had a fix buffer size for GIFDraw.
I use tft_espi lib.
I currently have an issue for setting the image position based on this example.
https://github.com/bitbank2/AnimatedGIF/blob/master/examples/TFT_eSPI_memory/GIFDraw.ino#L15

I don't understand where to put the X/Y offset.
I thought it was in setAddrWindow, but it's not working.

Sorry, it deviate from the original "bug" report.

bitbank2, 2022-02-20:
It would be best to provide me a complete sample project (including GIF image) that I can test. There are a million ways to make it crash that have nothing to do with the AnimatedGIF library. I'm highly doubtful that there is a bug in the library simply because it has been used in so many projects over the last two years (even with wider images). Changing the width limit doesn't fundamentally change anything except the amount of space reserved to hold a line's worth of pixels.

closed 2022-03-02 by bitbank2