bitbank2/AnimatedGIF #53: some gifs not rendering properly on hub75 panel using ESP32-HUB75-MatrixPanel-I2S-DMA library

h3llb3nt, 2022-03-18:
hello, trying to figure out why some of the gifs i attempt to display on my hub75 64x32 rbg matrix panel are coming up oddly. see the attached files. first is a working gif. second is a non-working test gif and third is video showing how it is rendering on my panel. i have not tested these tho on any other display yet (such as a pyportal as the original animatedGIF library examples seem to be based upon).

[image]

[image]

https://user-images.githubusercontent.com/22822528/159082362-cacdc553-879a-45f2-8feb-886e4800ac94.mov

i'm using code based on https://github.com/mrfaptastic/ESP32-HUB75-MatrixPanel-I2S-DMA/blob/master/examples/AnimatedGIFPanel/AnimatedGIFPanel.ino

bitbank2, 2022-03-18:
Thanks for providing a sample image. I'll get back to you shortly.

bitbank2, 2022-03-18:
I tested it and it works fine. The error is in the way that your GIFDraw callback is rendering each line. It appears to not be respecting the iX offset of each delta frame. See the example code in my library for how to do it correctly.

closed 2022-03-18 by bitbank2

jrlambs, 2024-03-01:
I'm having a very similar issue.  I suspect that it's this same thing... however I tried copying your GIFDraw method directly from the ESP32_LEDMatrix_I2S example and it does the same thing.  is there a specific example I should be pulling from for this type of gif?

jrlambs, 2024-03-01:
[GifPlayer.zip](https://github.com/bitbank2/AnimatedGIF/files/14466675/GifPlayer.zip)
Here's my applicaton

bitbank2, 2024-03-02:
I didn't write the LEDMatrix example. I suggest you try a simpler setup with an LCD to understand how the library works. Then you can modify the code for your LED matrix.