bitbank2/AnimatedGIF #54: Slow image refresh rate

Zaido5, 2022-04-29:
Sorry for putting this in the incorrect place before! Copying the statement:

"Thank you bit! I got it to work :) I am noticing that the frame refresh is quite slow, is there a way you recommend to speed it up? For reference I am using an ESP32.

also I had written a blurb about not getting the .h to work and I resolved that myself.

Thank you again! You are great!"

This is in reference to your [adafruit_gfx_memory.ino] code. Thank you in advanced, trying to learn as much as I can!

bitbank2, 2022-04-29:
What display are you using? What SPI frequency have you set? Can you share your code?

Zaido5, 2022-04-29:
[Stitch.txt](https://github.com/bitbank2/AnimatedGIF/files/8594768/Stitch.txt)
[Animate.txt](https://github.com/bitbank2/AnimatedGIF/files/8594793/Animate.txt)
[image]

Hi Bit! Attached are the files I'm using and the data for my ESP32. Thank you again sir!

bitbank2, 2022-04-30:
If your GIF images has lots of small runs of transparent pixels, it will cause it to slow down quite a bit. Take a look at my best_practices example code. It uses a RAM backing buffer to prevent this issue. Also have a look at my blog post about slow versus fast access to SPI LCDs:

https://bitbanksoftware.blogspot.com/2019/08/optimizing-access-to-serial-i2cspi.html

closed 2022-04-30 by bitbank2

Zaido5, 2022-04-30:
Ah, this is interesting. You are a true guru. I will do some more research and come back once I’ve gone through all the material. Thanks Bit!