bitbank2/AnimatedGIF #57: How can I skip frames faster in playFrame?

riraosan, 2022-08-03:
Hi. @bitbank2  I am @riraosan

I have been trying to find a way to draw a certain GIF animation at high speed by trial and error. However, I could not come up with a good idea. I have tried several fast rendering libraries, but I have to skip frames with frame waits less than 50ms.
How can I use a combination of playFrame and other APIs to skip frames with frame weights of 50ms or less at high speed?

Any advice would be appreciated.

Best Regards.

bitbank2, 2022-08-03:
Hi @riraosan,
You need to provide more info for me to help you. There are several strategies depending on the MCU and how much memory you have and what type of display you're using. For example, a worst case scenario is drawing frames with lots of transparent pixels directly onto a SPI LCD by repositioning the write pointer for each group of opaque pixels. With more info I'm sure I'll be able to help you.

riraosan, 2022-08-04:
Hi. @bitbank2 

Thank you for your reply and advice.

I am using the following in my personal project :

MCU: ESP32-PICO-D4(M5Stack ATOM Lite)
https://www.espressif.com/sites/default/files/documentation/esp32-pico-d4_datasheet_en.pdf

Digital TV : NTSC(CVBS)  (480i@29.97Hz)
Drawing library : LovyanGFX(M5GFX)
CVBS library : ESP32_8BIT_CVBS using [ESP_8_BIT_composite Library](https://github.com/Roger-random/ESP_8_BIT_composite)

My project
https://github.com/riraosan/ESP32_8BIT_CVBS/tree/master/sample/01_AnimatedGIF

GIF files
https://github.com/riraosan/ESP32_8BIT_CVBS/blob/master/sample/01_AnimatedGIF/data/non5.gif

# My story

## Objective

I want to synchronize background music and animated GIFs and play them on a digital TV. To achieve this, I want to skip GIF frames that have drawing delays at high speed. 

## My Problem
I could use LovyanGFX sprites to speed up the drawing very much. However, it took more than 50ms to draw one GIF animation frame with a WAIT of less than 50ms!

## My solution

I manually thinned out the frames below 50ms

I incorporated a program to thin out the frames that are delayed in rendering. (using the Ramer-Douglas-Peucker algorithm).

However, my idea did not work.
I can't control the fine delay, so I can't synchronize the background music and the GIF animation by any means.

## Other ideas

I thought about inlining the AnimatedGIF library like LovyanGFX, but I don't think I have the time or skills to do that, so I am second guessing myself.

This is my story. If you don't mind, can you give me some ideas?

I would appreciate any advice you could give me.

Thank you in advance.

bitbank2, 2022-08-04:
If I understand the problem, you're trying to simply speed up the GIF decoding itself, not the displaying of the image? My GIF code is quite optimized already and I doubt that "inlining" the code will get you any benefit. The speed of decoding a GIF frame is directly proportional to how 'complex' the image is. In terms of LZW compression, this means how many unique pixel patterns have to be encoded (as opposed to repeating patterns). In simpler terms, an image which looks like a photo will compress very badly and decode very slowly while an image with lots of constant color areas will compress well and decode quickly. 480i is a relatively large size image and it doesn't surprise me that a GIF frame can take more than 50ms to decode on the ESP32. There's not much you can do to fix this problem except to change the frame size to be smaller or make the image content 'simpler'. You can't skip frames unless each frame is completely independent and that defeats some of the size benefit of animated GIF. This may not be an option for you, but the ESP32-S3 executes code about 15% faster than the ESP32-Pico D4. So, your options:
- simplify the types of images you're encoding
- Use a smaller frame size
- Use a faster CPU
-

closed 2022-08-04 by bitbank2

riraosan, 2022-08-04:
Hi, @bitbank2 

I got some good ideas from you. 
I did think that it would be difficult to simply skip frames because of the decoding.

I hadn't thought of that idea. I was sticking to the current picture size. Your idea helped me.

I will measure the drawing time after reverting back to the original GIF animation picture size. And then I would like to revisit the specific measures.

I appreciate your assistance.

Thank you very much.

riraosan, 2022-08-06:
HI. @bitbank2 
It's done.

I ended up reducing the picture size. As a result, the GIF drawing time is shorter and I am able to fine tune the AV sync.
Thank you very much.
https://twitter.com/riraosan_0901/status/1555919578517360640?s=20&t=vFa028rg-ECdMa_7623h0A

riraosan, 2022-08-21:
https://youtu.be/LADcNpvA-8U
Finally, I was able to achieve this quality.
Thanks for your advise.