bitbank2/AnimatedGIF #60: Decode whole frame to RAM

androot2, 2022-12-28:
I try to use your library with Adafruit matrix (WS2812B). I found that writing each line separately and call matrix.show every time is very slow.
So, better practice will be decode full frame to RAM, but I can't implement it using your code, because callback function GIFDraw uses line instead of frame. How to modify that?

tobozo, 2022-12-28:
code is untested but this is what I'd do:

```cpp

uint16_t my_framebuffer[ max_width * max_height ]; 

void GIFDraw(GIFDRAW *pDraw)
{
  // use GIFDraw from example
  // comment out tft stuff, keep the logic
  // ( ... )
  // add this at the end
  int idx = pDraw->y * iWidth;
  memcpy( &my_framebuffer[idx], d, iWidth );
  if (pDraw->y == pDraw->iHeight -1 ) {
    // last scan line -> send frame buffer to led matrix
    WS2812B.writePixels( my_framebuffer, iWidth );
  }
}
```

bitbank2, 2022-12-28:
Tobozo ha the right idea; just copy each line into a full frame buffer as it's decoded. Easy peasy.

closed 2022-12-28 by bitbank2

androot2, 2022-12-29:
Thank you for inspiration. I modified it a little:

in GIFDraw I added at beggining:
`if (pDraw->y == 0) matrix->clear();`

and at the end:
` if (pDraw->y == pDraw->iHeight -1 ) matrix->show();`

And what I had earlier:

without transparency:
`matrix->drawRGBBitmap(pDraw->iX, y, usTemp, iWidth, 1);`

with transparency:
`matrix->drawRGBBitmap(pDraw->iX+x, y, usTemp, iCount, 1);`

I using <Adafruit_NeoMatrix.h> because, I have four, tiled matrixes.

tobozo, 2022-12-29:
if you're sending the whole framebuffer, is there a real need to `matrix->clear()` before updating?

setting all leds to black until every GIF scan lines are processed is likely going to produce a flicker effect

androot2, 2022-12-29:
I know, but if I send GIF only to the part of matrix the rest of that will be uncleaned from previous content.
There isn't any flickering:
matrix->clear() cleans only buffer
matrix->show() write buffer to LEDs

tobozo, 2022-12-29:
unless your matrixes have different sizes, that still looks like an unnecessary overhead since you overwrite every pixel during the decoding process anyway

does it become glitchy if you don't use matrix->clear() at all?