bitbank2/AnimatedGIF #62: Strange extremely long seek time with SPIFFS on ESP32C3 at the first frame

0x0fe, 2023-01-10:
While playing the same GIF in infinite loop from SPIFFS  the first frame of each loop is significantly slowed down to the point the loop is not smooth at all, i know this is not the gif itself because when played on windows it loops smoothly. At first i assumed it was due to the file reloading at each loop, but after digging i found this is caused solely by the seek call. I dont understand why but the first seek is extremely slow, in the order or 20ms, and subsequent seeks are very short, in the order of 50 to 300us. 
Do you know what could cause this and how to go around this issue?

[image]

[image]

bitbank2, 2023-01-10:
This is an interesting finding and not surprising about a FLASH based file system, but truly unrelated to AnimatedGIF. I'm sorry, but I can't fix SPIFFS to make animations smoother.

closed 2023-01-10 by bitbank2

0x0fe, 2023-01-10:
yes, it is not related to the internals of AnimatedGIF indeed, still i wonder why the call of this particular frame gets so long each time.