bitbank2/AnimatedGIF #64: FB datas in 1555 instead of 565

0x0fe, 2023-02-15:
Is there a way to format the FB datas in 1555 instead of 565 or 888? My target FB has alpha channel.

bitbank2, 2023-02-15:
The pixels are 8-bit with a 24-bit palette lookup. You can configure my existing code to deliver you the original palette and convert the pixels to 1555 yourself. Adding that option within the AnimatedGIF library doesn't make sense to me.

closed 2023-02-15 by bitbank2

0x0fe, 2023-02-15:
i see, but how to get the alpha channel (1bit) value? it is not contained in the 565 or 888.

bitbank2, 2023-02-15:
There is a single transparent color index per frame and it's part of the data passed to the draw callback (pDraw->ucTransparent). GIF doesn't support any form of alpha channel - it's an optional, single transparent color index.

0x0fe, 2023-02-15:
OK clear, yes, i should not call it alpha channel since it is only 1 bit transparent pixel.