bitbank2/AnimatedGIF #66: Issue using AnimatedGIFPanel

FM-silla, 2023-03-10:
Hi guys, im using AnimatedGIFPanel.ino example with an esp32s3 devkit C and a led matrix 64x32 P3. I was able to display the 64x32 gifs  after loading them on flash with spiffs but since yesterday I always get blank matrix. I think I've updated some libraries, ive tryed to rollback but nothing changed. Im able to run correctly the ESP32_LEDMatrix_I2S tho. Do you have any ideas? Thanks

bitbank2, 2023-03-10:
Hi @FM-silla . I'm not sure where you got AnimatedGIFPanel.ino, but it's not one of this repo's examples. I personally haven't written any of the LED matrix examples. Try using the return codes to diagnose the issue; errors are reported at each step.

closed 2023-03-10 by bitbank2

FM-silla, 2023-03-10:
> Hi @FM-silla . I'm not sure where you got AnimatedGIFPanel.ino, but it's not one of this repo's examples. I personally haven't written any of the LED matrix examples. Try using the return codes to diagnose the issue; errors are reported at each step.

OKs thanks i will try