bitbank2/AnimatedGIF #68: a example for tft_espi sd gif

hannescam, 2023-03-24:
it would be nice if a example for tft_espi gif from sd example would be provided

tobozo, 2023-03-24:
do you feel like creating it? 

you can take [this](https://github.com/bitbank2/AnimatedGIF/tree/master/examples/ESP32-LGFX-SDCard-GifPlayer) example as a template, and use the `GIFDraw()` function from  [this example](https://github.com/bitbank2/AnimatedGIF/blob/master/examples/TFT_eSPI_memory/GIFDraw.ino#L15).

bitbank2, 2023-03-24:
I think there are enough examples to get you going. The output of the GIFDRAW call is very simple - all you have to do is use the existing code to convert your pixels into RGB565/RGB888 and copy them to your output device one line at a time.

closed 2023-03-24 by bitbank2

hannescam, 2023-03-24:
but this example is for the m5 stack specific and was not meant for other boards mine is the esp32 capacitive touch camera from makerfabs

tobozo, 2023-03-24:
The use of M5 specifics is just a layer over standard components so it'll be fine.

I assume you already have a working TFT_eSPI configuration for the makerfabs device, so basically you only need to rename a few variables in the template sketch and comment out the specifics.

1) comment out the first include line and add yours:

```cpp
// #include <ESP32-Chimera-Core.h> // comment this out
#include <SD.h>
#include "YOUR_TFT_eSPI_Configuration.h"
#include <TFT_eSPI.h>
```

2) in the setup(), replace `M5.begin()` by `Serial.begin( 115200 );`
3) search every occurence of `M5STACK_SD` and replace it by `SD`, it was just an alias so it'll be fine.
4) comment out the `GIFDraw()` and `TFTDraw()` functions
5) copy the GIFDraw function from the other sketch
6) optional: implement esp_deep_sleep() to replace `M5.powerOFF();`

hannescam, 2023-03-24:
ok i will try that
and i have allready a working tft_espi config (over hspi) and a working sd config (also over hspi)

hannescam, 2023-03-24:
it doesnt work it doesnt find std::vector

tobozo, 2023-03-24:
> it doesnt find std::vector

adding `#include <vector>` or `#include <vector.h>` in your sketch should solve that

>  please provide me a functional sketch

I don't have your exact model of makerfabs board so I'm afraid the only help I can provide is theoritical.

hannescam, 2023-03-24:
so for the functional example i mean just for tft_espi with loading from sd
and i tried that with #include <vector> and it didnt work

tobozo, 2023-03-24:
I'm sorry I can't help you more than that, good luck in your research!

hannescam, 2023-03-26:
i managed to get it to compile and init both sd and TFT_eSPI successfully but it only displays a black screen with the serial output showing that it opens the file successfully (i combined the GIFDraw.ino from TFT_eSPI_memmory and the rest from adafruit_gfx_sdcard)
the code:

```
#include <AnimatedGIF.h>
#include <TFT_eSPI.h>
#include <SPI.h>
#include <SD.h>
#include <FS.h>

SPIClass spiSD(HSPI);

#define BUFFER_SIZE 256
#define BUILTIN_SDCARD 4

#ifdef USE_DMA
  uint16_t usTemp[2][BUFFER_SIZE]; // Global to support DMA use
#else
  uint16_t usTemp[1][BUFFER_SIZE];    // Global to support DMA use
#endif
bool     dmaBuf = 0;

TFT_eSPI tft = TFT_eSPI();
AnimatedGIF gif;
File f;

#define DISPLAY_WIDTH  tft.width()
#define DISPLAY_HEIGHT tft.height()

void * GIFOpenFile(const char *fname, int32_t *pSize)
{
  f = SD.open(fname);
  if (f)
  {
    *pSize = f.size();
    return (void *)&f;
  }
  return NULL;
} /* GIFOpenFile() */

void GIFCloseFile(void *pHandle)
{
  File *f = static_cast<File *>(pHandle);
  if (f != NULL)
     f->close();
} /* GIFCloseFile() */

int32_t GIFReadFile(GIFFILE *pFile, uint8_t *pBuf, int32_t iLen)
{
    int32_t iBytesRead;
    iBytesRead = iLen;
    File *f = static_cast<File *>(pFile->fHandle);
    // Note: If you read a file all the way to the last byte, seek() stops working
    if ((pFile->iSize - pFile->iPos) < iLen)
       iBytesRead = pFile->iSize - pFile->iPos - 1; // <-- ugly work-around
    if (iBytesRead <= 0)
       return 0;
    iBytesRead = (int32_t)f->read(pBuf, iBytesRead);
    pFile->iPos = f->position();
    return iBytesRead;
} /* GIFReadFile() */

int32_t GIFSeekFile(GIFFILE *pFile, int32_t iPosition)
{ 
  int i = micros();
  File *f = static_cast<File *>(pFile->fHandle);
  f->seek(iPosition);
  pFile->iPos = (int32_t)f->position();
  i = micros() - i;
//  Serial.printf("Seek time = %d us\n", i);
[... 191 lines cut ...]
  if (gif.open("/et.gif", GIFOpenFile, GIFCloseFile, GIFReadFile, GIFSeekFile, GIFDraw))
  {
    GIFINFO gi;
    Serial.printf("Successfully opened GIF; Canvas size = %d x %d\n", gif.getCanvasWidth(), gif.getCanvasHeight());
    if (gif.getInfo(&gi)) {
      Serial.printf("frame count: %d\n", gi.iFrameCount);
      Serial.printf("duration: %d ms\n", gi.iDuration);
      Serial.printf("max delay: %d ms\n", gi.iMaxDelay);
      Serial.printf("min delay: %d ms\n", gi.iMinDelay);
    }
    while (gif.playFrame(true, NULL))
    {
    }
    gif.close();
  }
  else
  {
    Serial.printf("Error opening file = %d\n", gif.getLastError());
    while (1)
    {};
  }
}
```

and i use platformio

hannescam, 2023-03-27:
and please reopen this issue

bitbank2, 2023-03-27:
This issue isn't a bug in AnimatedGIF and isn't really a feature request. I publish my own LCD library (bb_spi_lcd) that has plenty of examples in combination with this code. I don't use TFT_eSPI. You're free to add this to his examples if you think it's necessary.

hannescam, 2023-03-28:
ok