bitbank2/AnimatedGIF #69: Question, How better write text over the GIF animation ?

MIKHANYA, 2023-04-13:
Can You suggest How better write text over the GIF animation ? Maybe you know someone who has already implemented this functionality  ?

bitbank2, 2023-04-13:
You haven't given me much to go on. What target system? What type of images and text? Also, this isn't a valid issue to open here. It would be better to ask in a discussion forum such as Discord/Twitter/Mastodon/etc.

MIKHANYA, 2023-04-14:
> You haven't given me much to go on. What target system? What type of images and text? Also, this isn't a valid issue to open here. It would be better to ask in a discussion forum such as Discord/Twitter/Mastodon/etc.

Sorry for not giving you enough information. I use ESP32-PICO-D4 with screen ST7735 0.96-inch IPS(80 * 160). Used your code example TinyPico_Display_GIF.ino (with your GIF homer_car_small), replacing only the pin numbers. The idea is to keep the text (string, for example) above the animation.

mzashh, 2023-04-26:
i was wondering if you had figured it out, i tried doing it in void GIFDraw but the text always ends up flickring

bitbank2, 2023-04-26:
You would need to buffer the decoded image, draw the text over it and then push it to the display. You can do it on a line by line basis, but it's harder to code.

closed 2023-04-26 by bitbank2