bitbank2/AnimatedGIF #73: GIF Encoder

Borel-Kamnang, 2023-08-25:
Je recherche l'encodeur gif équivalent à faible mémoire.
do you think i can find it?
Indeed, I already had to try to readapt some encoders still found on Github but working on linux instead (tested on raspberry pi):

https://github.com/lecram/gifenc
https://github.com/charlietangora/gif-h

But it crashes my CPU. not knowing how to debug on arduino, I wonder if it is related to high memory allocation in the code well there is something else that escapes me

i am using ESP32 and Rapberry pico W

tobozo, 2023-08-25:
it's probably memory, difficult to say more without the actual crash message, image dimensions and frames amount :-)

if your ESP32 has PSRAM you can use `ps_malloc` instead of `malloc` and `ps_calloc` instead of `calloc`

Borel-Kamnang, 2023-08-25:
okay i will try that method. 
tank you.

Borel-Kamnang, 2023-08-30:
https://github.com/charlietangora/gif-h
work with ESP32 PS RAM memory allocation. 
I used ESP32-CAM

closed 2023-08-30 by bitbank2