bitbank2/AnimatedGIF #76: The ucLZW buffer access violaton

GlebPlekhotko, 2023-12-29:
Hello!

While playing with your library, I found, that for some reason the index into the _ucLZW_ array becomes greater than its actual size. It happens in the "_DecodeLZW_" function found in the "_gif.c_" file. The failure is hidden inside the _GET_CODE_ macro, so I have commented it out and replaced with its content manually. 

```
 // Main decode loop
 while (code != eoi && pImage->iYCount > 0) // && y < pImage->iHeight+1) /* Loop through all lines of the image (or strip) */
 {
     //GET_CODE

     if (bitnum > (REGISTER_WIDTH - codesize)) {
         pImage->iLZWOff += (bitnum >> 3);
         bitnum &= 7; ulBits = INTELLONG(&p[pImage->iLZWOff]); <== ERROR
     }

```

The _p_ buffer points to the _ucLZW_ buffer, which _LZW_BUF_SIZE_ (6 * 255 = 1530) bytes long, while the _iLZWOff_ reaches 1533, 1535, 1538 and so on. In my particular case, that results in the premature termination of the test GIF file decoding (only 54 lines of 400 available are decoded).

It felt like the library forgot to get more data from the file, so I traversed a code and found the following line in the _GIFMakePels_ function:

```
if ((pPage->iYCount & 3) == 0) // since we support only small images...
  ucNeedMore = 1;
```

I tried to replace 3 with 2 and everything started to behave as expected. The GIF file I fed was completely and successfully decoded.

Probably, the code was confused by the relatively complex GIF file I gave to it. Here it goes:

[image]

The problem was reproduced on both _gcc (MinGW)_ and _arm-none-eabi-gcc_ (Cortex-M4) toolchains.

Thanks for your attention.

bitbank2, 2023-12-31:
Thanks for finding this bug. I was trying to increase the performance, but a better solution would be to check that the pointer passes a "high water" mark instead of a random interval. BTW - your fix is not necessarily a good one because it will cause it to check 2 lines in a row and then skip 2 lines. A better plan might be to change the 3 to 1. I'll try a high water mark instead.

closed 2023-12-31 by bitbank2

bitbank2, 2023-12-31:
I just pushed new code (not a release yet). Appears to work fine.

GlebPlekhotko, 2024-01-09:
Thanks for spending your time. 

> your fix is not necessarily a good one because it will cause it to check 2 lines in a row and then skip 2 lines.

Actually I did not have a real display to ensure the fix works really fine. My primary interest was average time spent for decoding one line. Depending on the results, the final decision about future project progress will be committed.

By the way, if you are curious, I have compared your library with the venerable [giflib](https://giflib.sourceforge.net/).
The conditions were the following:
- LPC54605J512B (Cortex-M4);
- 180 MHz clock;
- arm-none-eabi-gcc version 9.2.1 20191025;
- -O2 optimization;
- test image stored in the internal flash;
- no software tweaks;

The average line (400 pixels) decoding time for the giflib was 644.8 microseconds, while your implementation took about 573.3 us. Of course, I was using the "fix" I noted above, so results might be a little different with the newest version.

Still presents a noticeable boost.

bitbank2, 2024-01-09:
I think the advantage of my code will be larger on certain CPUs, but I'll accept that. I don't think this change will affect it much.

GlebPlekhotko, 2024-01-09:
That is just for your information.
Maybe you're interested in feedback from the users running your code in the different environments. :-)

bitbank2, 2024-01-09:
I experimented with optimizing GIF for a long time. The true optimization comes when you make the output image become the dictionary. I have a version which does that and it runs much faster, but... it needs a LOT of RAM. This code is as fast as I can make it on a constrained device.