bitbank2/AnimatedGIF #79: Suggestion: gain a little more performance by engaging the non-aligned memory access.

GlebPlekhotko, 2024-01-10:
Hello!

In the end of the _AnimatedGIF.h_ file there are the following lines:
```
// Due to unaligned memory causing an exception, we have to do these macros the slow way
#define INTELSHORT(p) ((*p) + (*(p+1)<<8))
#define INTELLONG(p) ((*p) + (*(p+1)<<8) + (*(p+2)<<16) + (*(p+3)<<24))
```

Some architectures support the non-aligned memory access. The Cortex-M4 is one of them. You may find the approve in the [Generic User Guide](https://documentation-service.arm.com/static/5f2ac4ab60a93e65927bbdbf?token=), clause "_3.3.5 Address alignment_". 

So I tried to utilize this feature and altered those definitions:
```
// Due to unaligned memory causing an exception, we have to do these macros the slow way
#define INTELSHORT(p) (*((unsigned short int*)p))
#define INTELLONG(p)  (*((unsigned int*)p))
```

The test image was the same I attached [here](https://github.com/bitbank2/AnimatedGIF/issues/76). The image was decoded fine and about 5% faster. The average line decoding time for my test environment was 511.7 vs 541.6 us for the original "slow way".

Does this feature worth coming into your code? Of course it does not work for each and every CPU in the world, but at least Cortex-M4 as well as M3 are capable of it.

Thanks for your time.

closed 2024-01-10 by bitbank2

jrahlf, 2024-02-29:
ESP32-xx variants also supports unaligned access.

bitbank2, 2024-02-29:
I already added that to the latest code (see Turbo branch)