bitbank2/AnimatedGIF #82: GIFs look scrambled on 72x40 OLED Display

webfraggle, 2024-02-25:
**Describe the bug**
Tried to use the OLED example (gif_oled_demo) with a 72x40 OLED Displays and the GIF looks scrambled.

I tried this test.gif (and your test_images):
[image]

And it looks like this:
[image]

**To Reproduce**
Converted it to *.h file via your image_to_c Converter
Added it to the example
Changed size of displays in code
Changed OLED specs

The example text while starting the controller looks good, so I think it is no issue with the OLED library.

bitbank2, 2024-02-25:
Please zip of the complete example and share it here; small changes can cause things like this.

webfraggle, 2024-02-25:
[OLED_72x40_Gif-demo.zip](https://github.com/bitbank2/AnimatedGIF/files/14397115/OLED_72x40_Gif-demo.zip)

Here it is. 

Amazing fast response..Thank you.

bitbank2, 2024-02-25:
Nothing wrong with the GIF or my library, unfortunately I hard coded a value in my example that you copied into yours. This is what's causing the problem in the GIFDraw method:

index = x + ((y >> 3) << 7);

Your buffer is 72 bytes wide and the original demo is 128 bytes wide (1<<7).
Change it to:
index = x + ((y >> 3) * 72);

closed 2024-02-25 by bitbank2

webfraggle, 2024-02-25:
Many Many Thanks for the fast solution.