bitbank2/AnimatedGIF #83: Memory pointer GIFFILE::pData not const

jrahlf, 2024-02-29:
**Describe the bug**
GIFFILE::pData is `uint8_t*` instead of `const uint8_t*`

As far as I can tell, the memory is only read and never changed. Thus it should store a const pointer. 
Also `AnimatedGIF::open` should accept a const pointer then.

bitbank2, 2024-02-29:
For me, it doesn't matter if user-supplied pointers are const or not in the definition. Either way requires casting them in different situations.

closed 2024-02-29 by bitbank2

jrahlf, 2024-02-29:
It does matter whether `AnimatedGIF::open` takes `uint8_t*` or `const uint8_t*`:
1. Currently I don't know if AnimatedGIF modifies the data. Hopefully not.
2. If I have a `const uint8_t*` and pass it to `AnimatedGIF::open` **and** AnimatedGIF modified the data, this would be undefined behaviour. For example if the underlying data lies in read only section (e.g. flash memory).

Why would it require casting, if the pointer type would be const? It seems that the data is always copied from pData to other arrays.

bitbank2, 2024-02-29:
It can require casting if you allocate a non-const pointer, load data into it, then pass it to the library. The source data isn't modified.

jrahlf, 2024-02-29:
You can always pass a non-const `T*` to function taking a const `const T*`, so I do not think this is an issue.
Making the pointer const in the interface shall give the caller the guarantee, that the data is not modified.
```c++
void func(const int * param);
...
  int value;
  func(&value);
```