bitbank2/AnimatedGIF #87: turbot_t_qt_example is missing allocFrameBuf() call

thelastoutpostworkshop, 2024-06-08:
The example turbot_t_qt_example do not work, allocFrameBuf() is missing.

In the library DecodeLZWTurbo is expecting the frame buffer to be allocated :

    if (pImage->ucDrawType == GIF_DRAW_COOKED && pImage->pfnDraw && pImage->pFrameBuffer) { // convert each line through the palette

So the draw function is never called.

I added the frame buffer allocation in the example to make it work:
      if (gif.allocFrameBuf(GIFAlloc) != GIF_SUCCESS)
      {
        Serial.printf("allocFrameBuf failed\n");
        while (true)
          ;
      }

bitbank2, 2024-06-08:
I'm not sure if you have a bad copy of the repo or where you got that code. Here's what's currently on GitHub:

https://github.com/bitbank2/AnimatedGIF/blob/542d98dd2dd78769dc1c2aea70969f8f1cce65dd/examples/turbo_t_qt_example/turbo_t_qt_example.ino#L53

It allocates the turbo buffer and then passes it to the library with setTurboBuf().

thelastoutpostworkshop, 2024-06-08:
Yes I know, but the code of the example do not work if the frame buffer is not allocated, both of them must be allocated based on DecodeLZWTurbo in gif.inl 
if (pImage->ucDrawType == GIF_DRAW_COOKED && pImage->pfnDraw && pImage->pFrameBuffer)

bitbank2, 2024-06-08:
yes, sorry the turbo decoding needs the framebuffer for cooked output. I'll update the example.

bitbank2, 2024-06-08:
ok, pushed a fix

closed 2024-06-08 by bitbank2