bitbank2/AnimatedGIF #88: Loop count documentation and also does not work as expected

jrahlf, 2024-07-08:
**To Reproduce**
Try to play a GIF with a loop count (repeat count set).

**Expected behavior**
_Either_,
- the library should handle the loop by automatically resetting to the first frame and calculating the total number of frames (frames x loop count)
- or one should query the repeat count via `getLoopCount` and perform the looping outside of the library

**What actually happens**
1. The documentation is unclear if the library handles the looping or not. Apparently it does not. I think this should be added to the documentation.
2. `getLoopCount` always returns -1. The reason is, that `open` calls `GIFInit`, which calls `GIFParseInfo(..., bInfoOnly=1)`. This means the loop count information is not parsed. Unfortunately `getInfo` does not parse the field either, so currently it is not possible to get the loop count without modifying the library itself.

Please add a way to actually parse the loop count. Or even add loop handling to the library directly.

jrahlf, 2024-07-09:
seems to work for me

jrahlf, 2024-07-09:
I just figured, that `iRepeatCount` field is uint16_t, so actually it cannot hold -1. Instead the default value is set to UINT16_MAX then.
Maybe it makes sense to change this to int (32bit).

bitbank2, 2024-07-09:
I'lll correct that - thanks again.

closed 2024-07-09 by bitbank2