bitbank2/AnimatedGIF #90: GIF-image has pixel failures on TFT-Screen

JoJos1220, 2024-08-18:
**Describe the bug**
I got a problem in showing an .gif File (see below) on a TFT-screen with the TFT_eSPI.h libary.  The picture is decoded completly and is shown on the TFT Display. But there is a slight pixel-bar in the first third of the screen and some pixels which are not shown correctly. Also those pixels are shown outside of the picture and within the TFT-screen....

**To Reproduce**
I use this GifDraw() Function:

`
/*-----------------------------------------------------------------------------
 * GIFDraw() Function for Drawing an .gif Image to TFT Display
 * Callback-Funktionen for GIFDRAW Libary
 * Draw a line of image directly on the LCD
 * ----------------------------------------------------------------------------
*/
void GIFDraw(GIFDRAW *pDraw)
{
    uint8_t *s;
    uint16_t *d, *usPalette;
    uint16_t usTemp[SCREEN_WIDTH];
    int32_t  i, y, iWidth;

    iWidth = pDraw->iWidth;
    if (iWidth > SCREEN_WIDTH) {
        iWidth = SCREEN_WIDTH;
    }
    usPalette = pDraw->pPalette;
    y = pDraw->iY + pDraw->y; // current line
    
    s = pDraw->pPixels;

    if (pDraw->ucDisposalMethod == 2) { // restore to background color
        for (i = 0; i < iWidth; i++) {
            if (s[i] == pDraw->ucTransparent) {
                s[i] = pDraw->ucBackground;
            }
        }
        pDraw->ucHasTransparency = 0;
    }

    if (pDraw->ucHasTransparency) { // if transparency used
        uint8_t *pEnd, c;
        uint8_t ucTransparent = pDraw->ucTransparent;
        uint32_t  k = 0;
        uint32_t  iCount = 0; // count non-transparent pixels
        pEnd = s + iWidth;

        while (k < iWidth) {
            c = ucTransparent - 1;
            d = usTemp;

            // Sammeln und Zeichnen undurchsichtiger Pixel
            while (c != ucTransparent && s < pEnd) {
                c = *s++;
                if (c == ucTransparent) { // done, stop
                    s--; // back up to treat it like transparent
                } else { // opaque
                    *d++ = usPalette[c];
                    iCount++;
                }
            } // while looking for opaque pixels

            if (iCount) { // any opaque pixels?
                tft.setAddrWindow(pDraw->iX + x_offset + k, y + y_offset, pDraw->iX + x_offset + k + iCount - 1, y + y_offset);
                //tft.setAddrWindow(pDraw->iX + k, y, pDraw->iX + k + iCount - 1, y);
                tft.pushColors((uint16_t *)usTemp, iCount, true);

                k += iCount;
                iCount = 0;
            }

            // Überspringen transparenter Pixel
            c = ucTransparent;
            while (c == ucTransparent && s < pEnd) {
                c = *s++;
                if (c == ucTransparent) {
                    iCount++;
                } else {
                    s--; // back up to treat it like opaque
                }
            }

            if (iCount) {
                k += iCount; // skip these
                iCount = 0;
            }
        }
    } else { // No transparency
        s = pDraw->pPixels;
        for (i = 0; i < iWidth; i++) {
            usTemp[i] = usPalette[*s++];
        }
        tft.setAddrWindow(pDraw->iX + x_offset, y + y_offset, pDraw->iX + x_offset + iWidth - 1, y + y_offset);
        //tft.setAddrWindow(pDraw->iX, y, pDraw->iX  + iWidth - 1, y);
        tft.pushColors((uint16_t *)usTemp, iWidth, true);
    }
} /* GIFDraw() */
`

and this main-function to show the image on the TFT:

`
else if (_filename.endsWith(".gif")) {
        DEBUG_SERIAL.println("[Loop] Next GIF to show: " + _filename);

        if (gif.open(_path.c_str(), picOpen, picClose, gifRead, gifSeek, GIFDraw))
        {
            DEBUG_SERIAL.printf("Successfully opened GIF; Canvas size = %d x %d\n", gif.getCanvasWidth(), gif.getCanvasHeight());
            tft.startWrite();
            while (gif.playFrame(true, NULL))
            {
                yield(); 
            }

        tft.endWrite();
        gif.close();

        }
    }

`

**Expected behavior**
A clear picture on the screen without any pixel failure.

**Image files**
[image]

bitbank2, 2024-08-19:
It looks like you're using the setAddrWindow method incorrectly and this could be causing your incorrect pixel output.
The parameters are xs (starting x), ys (starting y), w (width), h (height). Your values for width and height are not correct.

closed 2024-08-19 by bitbank2

JoJos1220, 2024-08-27:
do you see my error in the function - actually?
I played now arround about 3 days and was not able to show the image propertly on the TFT-screen. I got always not a consistent picture with "ghost" pixels and so on....
x_offset/y_offset is "0"

JoJos1220, 2024-09-18:
Well, yes....it now worked for me.
I took again the example GIFDraw() function and needed to tft.setSwapBytes(TRUE) and it works now for me - thanks.

bitbank2, 2024-09-18:
Great - thanks for letting me know.