bitbank2/AnimatedGIF #96: how to determine the palette at runtime.

vindar, 2024-10-08:
Hi, 

Fist, I would like to thank you for all the libraries you are sharing with the community. I have been playing with some of them lately and they are great !  :-)

In my current project, I am using you AnimatedGIF library but I need to find out, from inside the `GIFDraw()` callback, which palette is currently available (i.e. is it `pDraw->pPalette` or `pDraw->pPalette24`) because I cannot always know in advance how `begin()` was called. I expected that the invalid palette pointer would be set to null but it seems that both pointers are always non-zero (even for the invalid palette). Is there a reason for this behaviour ? Is there any way to determine the palette pixel type at runtime ? 

Thanks !

bitbank2, 2024-10-08:
Both point to the same palette. It will be RGB565 if you request RGB565 as the output type and it will be RGB888 if you request RGB888 or RGB8888 as the output type.

closed 2024-10-08 by bitbank2

vindar, 2024-10-08:
Hi, 

Thanks for your answer. I understand that both pointers points to the same buffer, but with a different cast... My question is whether it is possible to *query* which palette is present (`RGB565` or `RGB888`) if you do not know which parameter was passed to `begin()` to start with. This information does not seem to be available in `GIFDRAW` which I expected to contain all information needed to perform the drawing in the callback (without having to rely on external parameters)  ?

bitbank2, 2024-10-08:
The information for the pixel/palette type is passed by the user at the initialization of the library. I currently don't have a method to query it because it shouldn't be necessary.

vindar, 2024-10-08:
Ok. Thank you for the confirmation. 

I ask this question because I am trying to implement bindings between your library and my graphic library [TGX](https://github.com/vindar/tgx/tree/improved-drawing-primitives). The user just has to add both libraries to his project and then TGX automatically binds around AnimatedGIF to let the user draw GIFs inside the  `tgx::Image` class.

To do so, I provide a generic `GIFDraw()` callback and create a template method that wraps around the `playFrame()` method of your library. However, I have no control over how the  user calls `begin()`  before all that so I do not know the palette type. There are workarounds to this problems but, none that is very elegant without a way to find the palette time at runtime.

I implemented similar bindings for your PNG and JPEG libraries and they works quite nicely (e.g. this [example](https://github.com/vindar/tgx/blob/improved-drawing-primitives/examples/Teensy4/2D/PNG_test/PNG_test.ino)) so I was hoping to do the same for GIF images to have a unified API...

bitbank2, 2024-10-08:
It's easy to get a pixelType() method or a GIFDRAW member variable. Which do you prefer?

vindar, 2024-10-09:
Hi, 

I just tested with several images and different palette type and it works fine :-) 

Thank you very much !