bitbank2/AnimatedGIF #99: how to use TFT_eSpi's TFT_eSprite to increase fps?

ham0nd, 2024-12-14:
hi , i use esp32 to display 240*240 gif with this library , but if the gif's file size too large ( > 6mb ),the fps is not good , and i saw the library's example only use TFT_eSpi's basic screen brushing method , so i think wether if can use TFT_eSpi's TFT_eSprite to increase fps?(i have enough ram to use) , thanks .

[image]

ham0nd, 2024-12-14:
Uploading VID_20241214_153140_20241214153420.mp4…

bitbank2, 2024-12-14:
There are three aspects to GIF decoding on the ESP32:
1) The decode speed of each line of 8-bit indexed pixels from the GIF file
2) The handling of the palette conversion, and transparent pixel handling
3) The display of the finalized pixels on the LCD
Ideally you want to push finalized pixels to the display in as large a group as possible to avoid pausing the data stream to send the LCD commands (e.g. set address window). The raw vs cooked option is helpful because it can manage the transparent pixels in RAM and push finished lines to the display. This however, requires more RAM to manage this option. You may find that changing your GIF file to make less use of transparent pixels will speed up display on the ESP32. There are really too many aspects to describe, but I suggest you learn more about how GIF files work.

closed 2024-12-14 by bitbank2