bitbank2/AnimatedGIF #100: Gifs glitching out

Neo0412, 2024-12-17:
Hey guys,

so i will try to keep it short.
I wanted to set up my CYD (ESP32-2432S028R) as an GIF Player and used the code of bitbank as reference.
The GIFs are located on the SD drive. 

So my problem is that most of my GIFs are showing some kind of graphical glitches as you can tell from the provided video in the ZIP file. I've provided you guys everything in the mentioned ZIP file (GIFs i am using, Code, Video of the problem)

I hope someone can help me with my issue. Thanks in advance! 
I am pretty new to C/C++ and ESPs but if you have any further questions about my project feel free to ask! 
[Issue.zip](https://github.com/user-attachments/files/18171313/Issue.zip)

bitbank2, 2024-12-18:
Thank you for submitting this issue. I am able to reproduce the problem on my equipment. I'll let you know when I have a fix.

Neo0412, 2024-12-19:
Hey,

quick update. It seems like some GIFs work. Especially smaller ones seem to work.
(See the GIFs i provided)

[working.zip](https://github.com/user-attachments/files/18202143/working.zip)

bitbank2, 2025-02-09:
I was able to trace through the code on the Mac to see the exact issue. It's that the GIFs use a local color palette and you cannot properly render that type of output unless you retain the entire frame as "cooked" pixels. The code you're using retains the 8-bit indexed pixels of the whole frame and only renders the current line as RGB565 output. When you use a global color palette, this can work to save memory. With a local color palette, the unchanged pixels (transparent pixels may be laid on top) do not change their fully rendered color output and this is where the problem lies. The only fix is to allocate a full framebuffer (8-bit) and a full framebuffer (16 or 24-bit) and don't use the GIFDraw callback. For your examples you'll need 320*180*3 bytes. Too large to allocate in static RAM on the ESP32, but could be kept in PSRAM.

closed 2025-02-09 by bitbank2

bitbank2, 2025-02-09:
As a conclusion, here's a video I made of the situation on an ESP32-S3:

https://youtu.be/0b_m-8fzq-A

The fix does not require any changes to AnimatedGIF. To properly display animations which use a local color palette, you need to allocate enough memory to hold the canvas as 8-bit indexed values AND the cooked output.

Neo0412, 2025-02-09:
Oh wow thank you very much for helping. I will try to fix my code now :)

bitbank2, 2025-02-10:
I just shared a new example which shows 3 ways to play animations and speed/memory tradeoffs for each:

https://github.com/bitbank2/AnimatedGIF/tree/master/examples/perf_explained

Neo0412, 2025-02-15:
Thanks again for your help.
Last question...
Will i be still able to play GIFs from my SD card? I heavily rely on Chat GPT to make this work so i was wondering if it was possible in the first place. Thanks!

bitbank2, 2025-02-16:
The AnimatedGIF library decodes image data into pixels. Providing that data is in your hands. I have various examples across my various imaging libraries which show how to read data from the SD card. If you're relying on ChatGPT to write your code, then I cannot help you unless you want to pay me for consulting. I can teach you how to program, but it will be less expensive to find this type of help someone else.