bitbank2/AnimatedGIF #103: Override MAX_WIDTH?

wystewart, 2025-01-19:
Would be really nice if MAX_WIDTH could be defined in the sketch or otherwise overridden without modifying AnimatedGIF.h.  This way the arduinoIDE/platformIO library manager can be used and updates made without having to modify .h file after update.

I'm using a 640x480 display on a STM32H750.

bitbank2, 2025-01-19:
Unfortunately in Arduino, you can't control macro definitions which get used to compile libraries unless I make the library a pure .H file. Hmm... I wonder if that would be a better way to go...
You may want to follow another project I'm working on -> https://github.com/bitbank2/pl_mpeg
I'm optimizing a MPEG-1 player and it will allow you to do video/animations with a lot smaller data compared to GIF or Motion-JPEG.

closed 2025-01-19 by bitbank2

wystewart, 2025-01-19:
Pretty sure there's a way to do it similar to have stm32sd includes a custom ffconf_custom.h if it exists in sketch.  Let me try and submit code to you if it works.  But note I can only test platformio arduino not the arduino ide

wystewart, 2025-01-19:
This seems to compile correctly 
#if __has_include("AnimatedGIF_custom.h")
  #include "AnimatedGIF_custom.h"
#endif

#ifdef __LINUX__
#define MAX_WIDTH 2048
#else
//#define MAX_WIDTH 320
#ifndef MAX_WIDTH
  #define MAX_WIDTH 480
#endif
#endif // __LINUX__

#define VALUE_TO_STRING(x) #x
#define VALUE(x) VALUE_TO_STRING(x)
#define VAR_NAME_VALUE(var) #var "="  VALUE(var)
#pragma message ("AnimatedGIF: " VAR_NAME_VALUE(MAX_WIDTH))

With   #define MAX_WIDTH 640 in AnimatedGIF_custom.h.

Compiles correctly in plaftormio arduino.

bitbank2, 2025-01-20:
Honestly, it seems easier to just edit AnimatedGIF.h instead of going through all of those gyrations. Or perhaps I can open the file buffer and line buffer to customization/dynamic allocation with a new API.

wystewart, 2025-01-21:
Prefer not to edit AnimatedGIF.h  so that I can upgrade the library and not have to worry about forgetting to edit MAX_WIDTH.  But I'll totally understand if you don't accommodate my request as  I can't really complain with this being a free library.

But here is a simpler way, how about this instead
#ifndef ANIMATEDGIF_MAX_WIDTH
  #define MAX_WIDTH 480
#else
  #define MAX_WIDTH ANIMATEDGIF_MAX_WIDTH
#endif

Then ANIMATEDGIF_MAX_WIDTH only need to be define as a compiler define option which is easy to do in Platformio.

bitbank2, 2025-01-21:
I like your new idea best.