bitbank2/AnimatedGIF #105: HTTP Stream to Decoder

VishBK, 2025-03-05:
Is there any way to give the decoder a Stream object so that the entire GIF file does not need to be stored in memory? I am trying to download a 128KB GIF from a URL on an ESP32 and display it on an LED matrix. Only getting the first frame of the GIF would also work.

closed 2025-03-05 by bitbank2

VishBK, 2025-03-09:
Thanks, ended up doing it by downloading the HTTP stream to a buffer with set size and using the callback functions to read from the buffer. Once the entire buffer is read, then fill up the buffer again from the stream.

aly-fly, 2025-03-17:
Another solution is to use WROVER instead of WROOM modules. These include PSRAM, easily available are 4 MB and 8 MB variants.  Most of the ESP32-S3 boards are available with PSRAM. Load full image into the RAM and let the library process the data from there. Example: 
https://github.com/aly-fly/RoomDisplay_S3/blob/main/src/ArsoRainGIF.cpp
https://github.com/aly-fly/RoomDisplay_S3/blob/main/src/GlobalVariables.cpp#L10